PlotJuggler/PlotJuggler #350: Sudden crash when using custom time series

ozzdemir, 2020-12-02:
When using custom time series, Plotjuggler crashes since lua interpreter continuously tries to run the function even if it is WIP (For example, while writing `return value-v1`, it crashes at `return value-v` since v does not exists) Stack trace is as follows, <WS_dir> is only a placeholder for the directory. I am using [PlotJuggler](https://github.com/facontidavide/PlotJuggler/commit/4e1b3e906cb8b043285a245a606e2b7a6a9c2ead) from source built in a catkin_ws. 
```
Stack trace (most recent call last):
#23   Object "", at 0xffffffffffffffff, in 
#22   Object "<WS_dir>/plotjuggler_ws/devel/.private/plotjuggler/lib/plotjuggler/plotjuggler", at 0x555740d0b3f9, in _start
#21   Source "../csu/libc-start.c", line 310, in __libc_start_main [0x7f8d0faa3bf6]
#20   Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/src/main.cpp", line 253, in main [0x555740d0a5d8]
        251:     w.show();
        252:     splash.finish(&w);
      > 253:     return app.exec();
        254:   }
        255:   MainWindow w(parser);
        256:   w.show();
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c6b9b3, in QCoreApplication::exec()
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c62909, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10cbd88e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#16   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4", at 0x7f8d0d9616db, in g_main_context_iteration
#15   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4", at 0x7f8d0d96164f, in 
#14   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4", at 0x7f8d0d961416, in g_main_context_dispatch
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10cbd4f0, in 
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10cbcd2d, in QTimerInfoList::activateTimers()
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c648d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5", at 0x7f8d11bee103, in QApplication::notify(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5", at 0x7f8d11be683b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c9407a, in QObject::event(QEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10ca0327, in QTimer::timerEvent(QTimerEvent*)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c9ffc6, in QTimer::timeout(QTimer::QPrivateSignal)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5", at 0x7f8d10c9366e, in QMetaObject::activate(QObject*, int, int, void**)
#4    Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/src/transforms/function_editor.cpp", line 715, in on_updatePreview [0x555740d7f1b8]
        712:       std::string name = new_plot_name.empty() ? "no_name" : new_plot_name;
        713:       PlotData& out_data = _local_plot_data.addNumeric(name)->second;
        714:       out_data.clear();
      > 715:       plot->calculate(_plot_map_data, &out_data);
        716: 
        717:       _preview_widget->removeAllCurves();
        718:       _preview_widget->addCurve(name, Qt::blue);
#3    Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/src/transforms/custom_function.cpp", line 99, in calculate [0x555740d7affc]
         96:   {
         97:     if (src_data.at(i).x > last_updated_stamp)
         98:     {
      >  99:       dst_data->pushBack(calculatePoint(src_data, channel_data, i));
        100:     }
        101:   }
        102: }
#2  | Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/src/transforms/lua_custom_function.cpp", line 55, in ~protected_function_result
    |    53:   new_point.x = old_point.x;
    |    54: 
    | >  55:   sol::safe_function_result result;
    |    56:   const auto& v = _chan_values;
    |    57:   // ugly code, sorry
      Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/./3rdparty/sol/sol.hpp", line 12382, in calculatePoint [0x555740d887a7]
       12379: 			err = call_status::runtime;
       12380: 		}
       12381: 		~protected_function_result() {
      >12382: 			stack::remove(L, index, popcount);
       12383: 		}
       12384: 	};
#1    Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/./3rdparty/sol/sol.hpp", line 6579, in __builtin_memcpy [0x555740d89285]
       6576: 		inline void remove(lua_State* L, int rawindex, int count) {
       6577: 			if (count < 1)
       6578: 				return;
      >6579: 			int top = lua_gettop(L);
       6580: 			if (top < 1) {
       6581: 				return;
       6582: 			}
#0    Source "<WS_dir>/plotjuggler_ws/src/PlotJuggler/3rdparty/lua-5.3.4/src/lapi.c", line 168, in lua_gettop [0x555740e63cf0]
        167: LUA_API int lua_gettop (lua_State *L) {
      > 168:   return cast_int(L->top - (L->ci->func + 1));
        169: }
Segmentation fault (Address not mapped to object [0x555700000020])

```

facontidavide, 2020-12-02:
Thanks for reporting, I will try to fix this ASAP

ozzdemir, 2020-12-02:
Sorry, I didn't see your edit. It happens to me all the time. What I do is import bag file and try to create the custom time series. It also happens when creating a new time series due to caching of the latest used function, as soon as I add a source time series.

facontidavide, 2020-12-04:
Which version are you using? I am not able to reproduce the error. Is there a sequence I can use to make the crash reproducible?

facontidavide, 2020-12-04:
I think it is fixed now. Check master

closed 2020-12-04 by facontidavide

ozzdemir, 2020-12-05:
Checked, and works flawlessly.
Thank you.

commit 4ae41182 (2020-12-04): bug fix #350 (crash in lua)