PlotJuggler/PlotJuggler #353: Compilation fails on OSX

Tobias-Fischer, 2020-12-09:
Hi there,

We are trying to compile plotjuggler for ROS on Conda (https://github.com/RoboStack/ros-noetic). We used to compile it just fine, however with the recent builds this error occurs for our OSX builds:

```
[30/306] Building CXX object 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/linux/FloatingWidgetTitleBar.cpp.o
FAILED: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/linux/FloatingWidgetTitleBar.cpp.o 
$BUILD_PREFIX/bin/x86_64-apple-darwin13.4.0-clang++ -DADS_STATIC -DCOMPILED_WITH_CATKIN -DPJ_MAJOR_VERSION=\"3\" -DPJ_MINOR_VERSION=\"0\" -DPJ_PATCH_VERSION=\"4\" -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DROS_BUILD_SHARED_LIBS=1 -I3rdparty/Qt-Advanced-Docking -I$SRC_DIR/ros-noetic-plotjuggler/src/work/3rdparty/Qt-Advanced-Docking -I3rdparty/Qt-Advanced-Docking/qt_advanced_docking_autogen/include -I$PREFIX/include -I$PREFIX/include/qt -I$PREFIX/include/qt/QtCore -I$PREFIX/./mkspecs/macx-clang -I$PREFIX/include/qt/QtWidgets -I$PREFIX/include/qt/QtGui -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I$PREFIX/include/qt/QtConcurrent -I$PREFIX/include/qt/QtPrintSupport -I$PREFIX/include/qt/QtXml -I$PREFIX/include/qt/QtWebSockets -I$PREFIX/include/qt/QtNetwork -I$PREFIX/include/qt/QtSvg -I$PREFIX/include/qt/QtOpenGL -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./include -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/color_widgets/include -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/Qt-Advanced-Docking/include -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/paho.mqtt.c/src -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/qwt/src -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/sol -I$SRC_DIR/ros-noetic-plotjuggler/src/work/./3rdparty/lua-5.4.1/src -I$SRC_DIR/ros-noetic-plotjuggler/src/work/3rdparty/Qt-Advanced-Docking/include/Qads -fdiagnostics-color=always -DBOOST_ERROR_CODE_HEADER_ONLY -g -fno-omit-frame-pointer  -fPIC -O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -mmacosx-version-min=10.9 -fPIC -std=c++14 -MD -MT 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/linux/FloatingWidgetTitleBar.cpp.o -MF 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/linux/FloatingWidgetTitleBar.cpp.o.d -o 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/linux/FloatingWidgetTitleBar.cpp.o -c $SRC_DIR/ros-noetic-plotjuggler/src/work/3rdparty/Qt-Advanced-Docking/src/linux/FloatingWidgetTitleBar.cpp
$SRC_DIR/ros-noetic-plotjuggler/src/work/3rdparty/Qt-Advanced-Docking/src/linux/FloatingWidgetTitleBar.cpp:197:34: error: too many arguments to function call, expected 0, have 1
                        d->FloatingWidget->showNormal(true);
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^~~~
$PREFIX/include/qt/QtWidgets/qwidget.h:482:10: note: 'showNormal' declared here
    void showNormal();
         ^
1 error generated.
```

Is it safe to just remove the `true` argument? Is there something which needs fixing here? I am not sure whether it's specific to the conda build.

Many thanks!

facontidavide, 2020-12-10:
Sorry, I don't maintain OSX. I am already busy enough with anything else :(

Tobias-Fischer, 2020-12-10:
No worries. I had a quick look, and it can be fixed with #361. It would be great if you could merge this PR :+1:

closed 2020-12-10 by facontidavide

tomjakubowski, 2020-12-11:
Hey, thanks for this! It seems like the commit merged here https://github.com/facontidavide/PlotJuggler/commit/0e7b6cc3100a69e838958f98190b191074f9fd7e got lost somehow. At least currently it's not reachable from main when I just cloned it, 2468863e31ab039bcd7fb3732f67aea20f1427a8.

2468863e31ab039bcd7fb3732f67aea20f1427a8

PR #361 (2020-12-10): Do not compile FloatingWidgetTitleBar on OSX
See https://github.com/githubuser0xFFFF/Qt-Advanced-Docking-System/blob/master/src/CMakeLists.txt#L52

Fix #353
PR #363 (2020-12-14): Apply #361 again to fix #353
Apply #361 again to fix #353

Commit got lost somehow ..

/cc @tomjakubowski