PlotJuggler/PlotJuggler #354: Cannot install on melodic due to conflicting packages

jcmonteiro, 2020-12-09:
Hi Davide,

I know that this issue is not exactly PlotJuggler's fault, but when installing the package it attempts to place the `MQTT*.h` headers from the 3rd party library https://github.com/eclipse/paho.mqtt.c in the root `/opt/ros/melodic/include` directory. In my use case, they conflict with another library that is using MQTT exactly like PlotJuggler - as a 3rd party package compiled from source.

A solution would be to edit the `CMakeLists` file and change the installation path, or have `libmosquittopp-dev` as a dependency for PlotJuggler instead of compiling from source.

What do you think?

facontidavide, 2020-12-09:
Ok, I will fix that ASAP, the truth is that we don't really need to install that at all.

Thanks for reporting

facontidavide, 2020-12-10:
should be fixed

closed 2020-12-10 by facontidavide