PlotJuggler/PlotJuggler #357: Compilation fails on Windows with Visual C++ 2017 since 3.0.4

gabm, 2020-12-10:
since commit https://github.com/facontidavide/PlotJuggler/commit/99d139470a1363cb6f9165c257f9ca0811785f19 the compilation fails on Visual C++ 2017 15.9.

There are some errors triggered by this

- `std::binary_function` has been removed in C++17 (deprecated in 11), but alphanum.hpp is still using it. Visual C++ reports an error here. I think its safe to remove as base class in this instance. I found more info here: https://stackoverflow.com/a/33115341
- sol doesn't compile with a strange bitset error (see below, translated to english)

```
[build] C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\include\bitset(92): error C2666: "operator /": 11 overloads have similar conversions [C:\Users\gabm\projects\os\PlotJuggler\build\src\plotjuggler.vcxproj]
[build]   c:\users\gabm\miniconda3\envs\plot-juggler\library\include\qt\qtgui\qmatrix4x4.h(126): note: could be "QMatrix4x4 operator /(const QMatrix4x4 &,float)"
[build]   C:\Users\gabm\miniconda3\envs\plot-juggler\Library\include\qt\QtGui/qquaternion.h(318): note: or "const QQuaternion operator /(const QQuaternion &,float)"
[build]   C:\Users\gabm\miniconda3\envs\plot-juggler\Library\include\qt\QtGui/qvector4d.h(281): note: or "const QVector4D operator /(const QVector4D &,const QVector4D &)"
[build]   C:\Users\gabm\miniconda3\envs\plot-juggler\Library\include\qt\QtGui/qvector4d.h(276): note: or "const QVector4D operator /(const QVector4D &,float)"
[build]   C:\Users\gabm\miniconda3\envs\plot-juggler\Library\include\qt\QtGui/qvector3d.h(281): note: or "const QVector3D operator /(const QVector3D &,const QVector3D &)"
[build]   C:\Users\gabm\miniconda3\envs\plot-juggler\Library\include\qt\QtGui/qvector3d.h(276): note: or "const QVector3D operator /(const QVector3D &,float)"

(continues several lines more)
```

facontidavide, 2020-12-10:
Thanks for reporting. I will have a look

facontidavide, 2021-01-24:
fixed

JimmyDaSilva, 2021-01-25:
Sorry @facontidavide but I still get the same error (mentioned above: 'bitset' and '/') with Visual C++ 2017 15.9, Qt 5.14.2 ( built with MSVC2017 64bit) and of course the latest PlotJuggler commit

The issue is coming from `function_editor.cpp`

facontidavide, 2021-01-25:
can you be more specific?

JimmyDaSilva, 2021-01-25:
I wish I could...
I am terrible on Windows and Visual Studio. From this, the only thing I got is that the error comes from this bitset issue in `function_editor.cpp`

```
2>------ Build started: Project: plotjuggler, Configuration: RelWithDebInfo x64 ------
2>Automatic MOC and UIC for target plotjuggler
2>function_editor.cpp
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\bitset(92): error C2666: 'operator /': 11 overloads have similar conversions
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qmatrix4x4.h(126): note: could be 'QMatrix4x4 operator /(const QMatrix4x4 &,float)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qquaternion.h(319): note: or       'const QQuaternion operator /(const QQuaternion &,float)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector4d.h(282): note: or       'const QVector4D operator /(const QVector4D &,const QVector4D &)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector4d.h(277): note: or       'const QVector4D operator /(const QVector4D &,float)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector3d.h(282): note: or       'const QVector3D operator /(const QVector3D &,const QVector3D &)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector3d.h(277): note: or       'const QVector3D operator /(const QVector3D &,float)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(236): note: or       'double operator /(int,qfloat16) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(236): note: or       'double operator /(qfloat16,int) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(226): note: or       'float operator /(float,qfloat16) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(226): note: or       'float operator /(qfloat16,float) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(225): note: or       'double operator /(double,qfloat16) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(225): note: or       'double operator /(qfloat16,double) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(224): note: or       'long double operator /(long double,qfloat16) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(224): note: or       'long double operator /(qfloat16,long double) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qfloat16.h(207): note: or       'qfloat16 operator /(qfloat16,qfloat16) noexcept'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector2d.h(257): note: or       'const QVector2D operator /(const QVector2D &,const QVector2D &)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qvector2d.h(252): note: or       'const QVector2D operator /(const QVector2D &,float)'
2>f:\qt\5.14.2\msvc2017_64\include\qtgui\qtransform.h(412): note: or       'QTransform operator /(const QTransform &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qsize.h(375): note: or       'const QSizeF operator /(const QSizeF &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qsize.h(200): note: or       'const QSize operator /(const QSize &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qmargins.h(427): note: or       'QMarginsF operator /(const QMarginsF &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qmargins.h(213): note: or       'QMargins operator /(const QMargins &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qmargins.h(207): note: or       'QMargins operator /(const QMargins &,int)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qpoint.h(407): note: or       'const QPointF operator /(const QPointF &,qreal)'
2>f:\qt\5.14.2\msvc2017_64\include\qtcore\qpoint.h(208): note: or       'const QPoint operator /(const QPoint &,qreal)'
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\bitset(92): note: or       'built-in C++ operator/(size_t, std::bitset<64>::<unnamed-enum-_Bitsperword>)'
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\bitset(92): note: while trying to match the argument list '(size_t, std::bitset<64>::<unnamed-enum-_Bitsperword>)'
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\bitset(91): note: while compiling class template member function 'bool std::bitset<64>::_Subscript(size_t) const'
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\bitset(392): note: see reference to function template instantiation 'bool std::bitset<64>::_Subscript(size_t) const' being compiled
2>f:\plotjuggler\3rdparty\sol\sol.hpp(10247): note: see reference to class template instantiation 'std::bitset<64>' being compiled
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\type_traits(616): note: see reference to class template instantiation 'std::basic_string_view<char,std::char_traits<char>>' being compiled
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xstring(2124): note: see reference to class template instantiation 'std::is_convertible<const _StringViewIsh (&),std::basic_string_view<char,std::char_traits<char>>>' being compiled
2>        with
2>        [
2>            _StringViewIsh=char [2]
2>        ]
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xstring(2122): note: see reference to variable template 'const bool conjunction_v<std::is_convertible<char const (&)[2],std::basic_string_view<char,std::char_traits<char> > >,std::negation<std::is_convertible<char const (&)[2],char const *> > >' being compiled
2>f:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xstring(3774): note: see reference to alias template instantiation 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>::_Is_string_view_ish<_StringViewIsh>' being compiled
2>        with
2>        [
2>            _StringViewIsh=char [2]
2>        ]
2>Done building project "plotjuggler.vcxproj" -- FAILED.
```

facontidavide, 2021-01-25:
Mhhh, the actual problem seems to be something in sol.hpp, probably related to C++17 support. I need to investigate.

It compiles fine with Visual Studio 2019

JimmyDaSilva, 2021-01-25:
Good to know that VS2019 works. I will install this version too then.

JimmyDaSilva, 2021-01-25:
Indeed it does work with VS2019. Add the color issue is fix as expected. Thanks.

facontidavide, 2021-05-18:
I just tried compiling on VS2017 and it works

closed 2021-05-18 by facontidavide