PlotJuggler/PlotJuggler #359: Apply filter disabled after bag file restarts.

ozzdemir, 2020-12-10:
I was working with a bag file in loop mode so that it restarts after bag file ends and observed that when that happens, applied filters were disabled. When I tried to check the "apply filter to data" menu, I observed that data sources were not visible. When I tried to re-apply the transforms, PlotJuggler crashed with the following stack trace. Note that, this is observed only when /use_sim_time param was true. I was able to reproduce the same with using bag files in datasamples with the stages below

I also tried to record the reproduction, video can be seen in [pj-bug.zip](https://github.com/facontidavide/PlotJuggler/files/5673314/pj-bug.zip)

Steps to reproduce:
* rosparam set /use_sim_time true
* rosbag play tilt_laser.bag -l --clock
* apply custom transform to a signal
* wait for bag file to restart (results in negative time instances)

```
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9f178f16, in QDialog::exec()
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e00a909, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e06588e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#28   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbe9b55d6db, in g_main_context_iteration
#27   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbe9b55d64f, in g_main_context_dispatch
#26   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbe9b55d416, in g_main_context_dispatch
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7fbe9712425f, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fbe9e5a72ea, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fbe9e5d0054, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fbe9e5ce582, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e00c8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9ef96103, in QApplication::notify(QObject*, QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9ef8e83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9efea7d9, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9efe816a, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9ef95631, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e00c8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9ef9665e, in QApplication::notify(QObject*, QEvent*)
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9ef8e814, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e00c5bb, in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9f1e1b8a, in QAbstractItemView::viewportEvent(QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9f06e4dd, in QFrame::event(QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9efcd99e, in QWidget::event(QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9f1dcb38, in QAbstractItemView::mousePressEvent(QMouseEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbe9f213c81, in QListView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9dfc0092, in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9dfbd4d4, in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9dfb8626, in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&)
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e03b554, in QMetaObject::activate(QObject*, int, int, void**)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbe9e03b618, in QMetaObject::activate(QObject*, int, int, void**)
#1  | Source "/home/<$USER>/WSs/plotjuggler_ws/build/plotjuggler/src/plotjuggler_autogen/EWIEGA46WW/moc_plotwidget_transforms.cpp", line 125, in DialogTransformEditor::qt_metacall(QMetaObject::Call, int, void**)
    |   123:     if (_c == QMetaObject::InvokeMetaMethod) {
    |   124:         if (_id < 5)
    | > 125:             qt_static_metacall(this, _c, _id, _a);
    |   126:         _id -= 5;
    |   127:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
      Source "/home/<$USER>/WSs/plotjuggler_ws/build/plotjuggler/src/plotjuggler_autogen/EWIEGA46WW/moc_plotwidget_transforms.cpp", line 124, in DialogTransformEditor::qt_metacall(QMetaObject::Call, int, void**) [0x55e40fb957c3]
        121:     if (_id < 0)
        122:         return _id;
        123:     if (_c == QMetaObject::InvokeMetaMethod) {
      > 124:         if (_id < 5)
        125:             qt_static_metacall(this, _c, _id, _a);
        126:         _id -= 5;
        127:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
#0    Source "/home/<$USER>/WSs/plotjuggler_ws/src/PlotJuggler/src/plotwidget_transforms.cpp", line 162, in DialogTransformEditor::on_listTransforms_itemSelectionChanged() [0x55e40fb30ad4]
        159:   QString transform_ID = selected_transforms.front()->text();
        160: 
        161:   auto curve_it = _plotwidget->curveFromTitle(curve_name);
      > 162:   auto qwt_curve = curve_it->curve;
        163:   auto ts = dynamic_cast<TransformedTimeseries*>( curve_it->curve->data() );
        164: 
        165:   QSignalBlocker block(ui->lineEditAlias);
Segmentation fault (Address not mapped to object [0x20])
```

facontidavide, 2020-12-10:
wow, this looks nasty. I will have a look

facontidavide, 2020-12-10:
this particular bug has been fixed, but I found a new one thanks to you... thanks !!!

closed 2020-12-10 by facontidavide