PlotJuggler/PlotJuggler #360: White font on white background is difficult to read

romainreignier, 2020-12-10:
Hi @facontidavide and thank you for this new Plotjuggler 3.0.

I was amazed to see that it has already reached OpenRobotics repos and is already installed in my system, just after the post on Discourse!

In the *Streaming* panel the choices in the combobox appear with a white font on white background when hovered on my system which makes the one hovered unreadable.
I use Kubuntu (KDE) 18.04 without any fancy theme, just the default one.

See:
[image]

I have tried the version 3.0.2 from the repos and 3.0.4 from `main` branch (you should change the branch name in the issue template, it stills uses `master` wording).

facontidavide, 2020-12-10:
That is weird and, as you may imagine, this is not what happens on my computer ;) 

This is probably due to some specific theme settings in your Kubuntu  that need to be overridden.

Since I can not replicate this on my side, I need your help to fix it.

[stylesheet_light.zip](https://github.com/facontidavide/PlotJuggler/files/5674452/stylesheet_light.zip)

Download and uncompress this file.

Try to load it with the command **Tools->Load Stylesheet** and tell me if it solves the issue.

facontidavide, 2020-12-10:
I am confident it is ok now

romainreignier, 2020-12-10:
Indeed, it works now, I can see the labels.
But there is not effect on hovering. I don't know if it is desired.

facontidavide, 2020-12-10:
no, it is not derired :(

facontidavide, 2021-05-16:
I am closing this. Let me know if with the latest version the problem persists.

closed 2021-05-16 by facontidavide