PlotJuggler/PlotJuggler #362: [Feature Request] Save all plots to file

kikass13, 2020-12-11:
Hi,

Maybe Im dumb, but as far as I can see it, there is no way to save plots to file. I can save a single plot to a file, but not the whole view (a single graph is not really helpfull). I mean i could do screenshots, but these scale badly and i have to manually cut out the not important parts of the plots.

facontidavide, 2020-12-11:
You are right.

There was originally a way to do that in PlotJuggler 2.x, but it is not ported yet to PJ 3.x

It will take me a while before I can work on this.

edward9503, 2022-09-07:
Hi @facontidavide is this feature already applied? And how to save multiple images into vector graphics at once?
 Thanks

andrestoga, 2024-01-16:
Yeah, I'm also wondering if this feature is already in main.

@facontidavide

TareqAlqutami, 2024-02-08:
It would be great if this feature is available.

khushal-wheel, 2024-07-18:
This would be super helpful! Plotjuggler is awesome!

GauravKumar9920, 2026-02-01:
@facontidavide What's the status of this Issue, have there been any updates ? 
I would love to work on this !

andrestoga, 2026-02-01:
> [@facontidavide](https://github.com/facontidavide) What's the status of this Issue, have there been any updates ? I would love to work on this !

@GauravKumar9920  It's already implemented! See [here](https://github.com/facontidavide/PlotJuggler/pull/1118)

closed 2026-02-01 by facontidavide