PlotJuggler/PlotJuggler #364: build from source

DimitriEckert137, 2020-12-14:
Ok maybe this is a stupid question but hopefully you can help me.
I stumbeld across the same problem as described [here](https://github.com/facontidavide/PlotJuggler/issues/350#issue-755162397)
This seems to be fixed in 3.0.5 which is not yet available via apt (right?). So I tried to install from source according to the instructions [here](https://github.com/PlotJuggler/plotjuggler-ros-plugins). This seemed to work all well but when I start it with `roslaunch plotjuggler_ros plotjuggler.launch ` I get this error message: 
```
Missing package [plotjuggler-ros]
 If you just upgraded from PlotJuggler 2.x to 3.x , try installing this package:

sudo apt install ros-${ROS_DISTRO}-plotjuggler-ros
```
But plotjuggler_ros seems to have built properly? And `sudo apt install ros-${ROS_DISTRO}-plotjuggler-ros` says:
 `ros-melodic-plotjuggler-ros is already the newest version (1.0.0-2bionic.20201202.150027)`
So I have either the apt version which works but has the bug or the source version which has the bug fixed but has some other issue.
Is there a way to get around this?

facontidavide, 2020-12-14:
Try 

         rosrun plotjuggler plotjuggler

DimitriEckert137, 2020-12-14:
When I do 
`     rosrun plotjuggler plotjuggler`
outside of the ~/ws_plotjuggler workspace it starts the older version (3.0.3) of plotjuggler where the bug is not yet fixed.
When I do 
`     rosrun plotjuggler plotjuggler`
within the ~/ws_plotjuggler workspace I get the same error message as above. Maybe I should clarify when I cancel the error message plotjuggler still starts but I cant load any rosbags into it.

facontidavide, 2020-12-15:
can you please share the output of your console when you launch the application?

ozzdemir, 2020-12-15:
@DimitriEckert137 , did you try closing the error message? I also see that message, but when I close it, plotjuggler starts and works normally (in my experience).

DimitriEckert137, 2020-12-20:
sorry for the late reply, was crazy busy the last days.
when i run 
` rosrun plotjuggler plotjuggler`
outside the ~/ws_plotjuggler workspace the console output is as follows:
```
Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamMQTT.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/home/dimitri/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler_ros"
"libDataLoadROS.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Number of plugins loaded:  4 

```

When i run it from within the sourced workspace the console output is the following (after clicking on cancel on the above mentioned popup error message):
```
Loading compatible plugins from directory:  "/home/dimitri/ws_plotjuggler/devel/.private/plotjuggler/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamMQTT.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/home/dimitri/.local/share/PlotJuggler"
Number of plugins loaded:  0 
```

facontidavide, 2020-12-24:
If you run the command

     source ~/ws_plotjuggler/devel/setup.bash

This is the expected output:

[image]

DimitriEckert137, 2020-12-25:
Ok. I checked and i can see that the directories in your output do exist in my workspace. but somehow it takes the one printed in the output i posted above. 
I am not too good at these things so i don't know what the problem could be. Could it be because the version from apt is still installed? Do I need to deinstall the old version first?

ovaag, 2020-12-26:
I have this same issue, tried with both ros noetic and melodic to build plotjuggler 3 from source. Building the version-2.x branch woroks as expected and is able to load ros-plugin with `rosrun plotjuggler plotjuggler`. Seem to be some issue with the new version of plotjuggler

facontidavide, 2021-01-05:
I worked on this issue, but I am not sure I fixed it (probably not).
32e89189f7efaf2c4333057ab8a9ac2493a6e036

I suppose that the correct path for your specific case is **/root/ws_plotjuggler/devel/.private/plotjuggler_ros/lib/plotjuggler_ros**, but double check!!!

The solution for you is to add manually the folder with the command:
   
     rosrun plotjuggler plotjuggler --plugin_folders=<your-folder-full-path>

closed 2021-01-05 by facontidavide

facontidavide, 2021-01-05:
I added a new feature you will love ;) 

[image]