PlotJuggler/PlotJuggler #365: mdf support

huckl3b3rry87, 2020-12-15:
I am curious is there any interest in supporting visualization mdf (Media Descriptor File)?

facontidavide, 2020-12-16:
Interesting. Is there any c++ library you are aware of to parse this format?

facontidavide, 2020-12-16:
I see, it looks like I can use this. https://www.turbolab.de/mdf_libf.htm

This is interesting, but since I am not directly interested in this, it might take a while (or a [sponsor](https://github.com/sponsors/facontidavide)) before I consider working on this feature.

Bartimaeus-, 2022-07-03:
@facontidavide I am also interested in MDF file support. This would greatly expand PlotJuggler's utility for analyzing CAN bus telemetry. While there's a plugin for the socketCAN log file format already, MDF is more widely used and is often the format that you'd export to from expensive tools from Kvaser, Vector, etc. in order to view the data with other tools. 

There is a Python tool called asammdf that has a GUI plotter for MDF files, but imo it is nowhere near as nice to use as PlotJuggler and I think there would be a lot of  asammdf GUI users that would adopt PJ  https://github.com/danielhrisca/asammdf

Here are some libraries that I've come across while looking into this:
https://github.com/R1tschY/libmdf4
https://github.com/CSS-Electronics/mdf4-converters
https://bitbucket.org/tobylorenz/asam_mdf/src/master/

The MDF loader would be most helpful if it functioned similar to the [DataLoadCAN plugin](https://github.com/PlotJuggler/plotjuggler-CAN-dbs/tree/development/DataLoadCAN) (using an optional .dbc file to parse signals) and worked on Windows and Linux

facontidavide, 2022-07-03:
Sorry, I have already a number of tasks in my backload and all of them are sponsored tasks.
I don't think that I will be able to work on this feature.

facontidavide, 2026-10-01:
PJ4 now has an official **MF4/MDF Loader**, including optional DBC decoding for CAN channels. Install it from the PJ4 marketplace. The [plugin documentation](https://github.com/PlotJuggler/pj-official-plugins/tree/main/data_load_mf4) describes supported formats and limitations. Closing this request as implemented in PJ4.

closed 2026-10-01 by facontidavide