PlotJuggler/PlotJuggler #366: Identity of transforms from /tf are swiched during plotting

rasmusan, 2020-12-16:
When subscribing to /tf or /tf_static with PlotJuggler in ROS2, the available transforms are listed as:
transforms.0
transforms.1
transforms.2
...

If multiple publishers publish to /tf, the identity of the numbered transforms in PlotJuggler changes. This causes plots from /tf to be useless. The id of each transform is specified as a string in the message (child_frame_id), but this is unavailable in PlotJuggler. Therefore there doesn't seem to be any way to "recover" a specific tf for plotting.

It would really be usefull with a way to filter messages based on child_frame_id. Some ideas comes to mind;

1) Support strings for custom series creation, and make it possible to "match" from an array based on a string.

2) Make "special" handling of tf's, as this is so common in ROS. Name Transform instances based on child_frame_id instead of based on array indices:
transforms.base
transforms.elbow
transforms.tcp
...

In general; thanks for this really great tool! Especially ROS2 support is appreciated.

For reference, I'm using
- PlotJuggler 3.0.5
- plotjuggler-ros-plugins commit 92abaa36
- ROS2 Foxy

facontidavide, 2020-12-16:
this is weird, because I am pretty sure that what you mention is already implemented. Can you share a rosbag to reproduce the issue?

rasmusan, 2020-12-16:
Yes of course. I have uploaded a rosbag and how it looks after loading into PlotJuggler here. I have plotted a few series of translation data to visualize the problem.

[rosbag2_2020_12_16-09_50_41.zip](https://github.com/facontidavide/PlotJuggler/files/5701421/rosbag2_2020_12_16-09_50_41.zip)
[image]

facontidavide, 2020-12-18:
problem fixed

closed 2020-12-18 by facontidavide