PlotJuggler/PlotJuggler #367: "STMStudio like" input plugin

martonmiklos, 2020-12-20:
Hey folks!

I am using a tool called STMStudio for debugging ARM microcontrollers made by ST. 

This tool basically parsing the elf files DWARF section with GDB to create a list from the used variables. (Btw. it smells GPL violation because they using a modified gdb without source code publication)

Then these extracted variables could be added onto several types of graphs and could be monitored in real time through the ST's debugger hardware. 

For the later part (streaming variables from the target) we could use the openocd which would allow to use this solution with wide variety of microcontrollers (including different vendors and architectures).

I opened this issue to host some brainstorming/idea exchange in this topic because I might spend some time on implementing something similar once I manage to allocate some free time to this.

facontidavide, 2020-12-20:
Be my guest!

facontidavide, 2021-01-06:
How is it going? I guess this is your fork https://github.com/Rescube/PlotJuggler

martonmiklos, 2021-01-07:
Thanks for asking!

You are correct this is the fork on what I am working on.

I did some investigation on what and how ST implemented in their STM studio.

Basically they using the GDB to extract the variables and their memory address and type from the ELF file.
I have implemented the same approach in  Qt/C++. Performance wise it has some are of improvement, but the general approach (using an external tool for elf parsing) seems feasible for me.

As a next step I will create a treeview and a model for displaying the available variables.
The current elf parsing takes a while and I would like to improve it by a couple of things:
 - Using the MI (machine interface) of the GDB instead of the command line parsing (I am not sure what I will gain with it)
 - The current variable parsing has some area of improvements (size/type info of all array items are queried for e.g.), and I might make sense to query the top level variables only first and then query the informations from the GDB (array/struct members) as the user opens the variables in the treeview.

This ELF parsing and UI is just a small part of the story. I will have need to figure what is the best way to dump the selected memory regions (i.e. variables) continuously with background memory access via openOCD. My initial plan is to use the openOCD TCP socket interface and use the memory access commands.

facontidavide, 2026-10-01:
Closing this PJ3 report about an STMStudio/OpenOCD input plugin during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide