PlotJuggler/PlotJuggler #368: Windows version 3 executable with UDP streaming

seanmcleod, 2020-12-21:
I'm busy adding a UDP streaming option to [JSBSim](https://github.com/JSBSim-Team/jsbsim) so that PlotJuggler can be used in real-time to view telemetry from JSBSim simulations.

I downloaded and installed the [Windows 2.8.4 Installer](https://github.com/facontidavide/PlotJuggler/releases/download/2.8.4/PlotJuggler_WIN_Installer-2.8.4.exe) but it doesn't appear to come with the UDP streaming plug-in?

The main readme for PlotJuggler has been updated to refer to version 3 of PlotJuggler but there doesn't appear to be a Windows build for version 3?

Looking at the commits and the results of from AppVeyor it doesn't look there has been a successful Windows build of version 3 yet?

facontidavide, 2020-12-21:
Indeed, I am a bit late with the work to be done to compile PlotJuggler3 on Windows.

You can give it a try and compile from source. I guess there are only minor errors :wink:

jlmcmchl, 2020-12-30:
I took a stab at building for windows under MSYS2, following the details from #315 - I wasn't able to build the LSL plugin or libraries, but the rest successfully built. Unfortunately, when dragging data into a plot widget it crashes. Threw it in gdb and found the following crash on startup:

```gdb
(gdb) r
Starting program: H:\Msys2\home\Jordan McMichael\PlotJuggler\install\bin\plotjuggler.exe
[New Thread 1468.0x4cf0]
[New Thread 1468.0x7924]
[New Thread 1468.0x7898]
[New Thread 1468.0x8250]
[New Thread 1468.0x4ffc]
[New Thread 1468.0x7b70]
[New Thread 1468.0x4a58]
[New Thread 1468.0x7d28]
[New Thread 1468.0x3678]
[New Thread 1468.0x7d08]
[New Thread 1468.0x561c]
[New Thread 1468.0x7808]
[Thread 1468.0x7808 exited with code 0]
[New Thread 1468.0x6764]
[New Thread 1468.0x64d8]
[New Thread 1468.0x1768]

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000502474 in QwtPlotAbstractGLCanvas::setLineWidth (this=0xe1d2f30,
    width=2)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_abstract_canvas.cpp:1027
1027        if ( width != d_data->lineWidth )
(gdb) bt
#0  0x0000000000502474 in QwtPlotAbstractGLCanvas::setLineWidth (
    this=0xe1d2f30, width=2)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_abstract_canvas.cpp:1027
#1  0x0000000000504c2a in QwtPlotOpenGLCanvas::qt_static_metacall (
    _o=0xe1d2f00, _c=QMetaObject::WriteProperty, _id=2, _a=0x58fd4a0)
    at PlotJuggler/build/3rdparty/qwt/src/qwt_static_autogen/include/moc_qwt_plot_opengl_canvas.cpp:135
#2  0x0000000000504fa0 in QwtPlotOpenGLCanvas::qt_metacall (this=0xe1d2f00,
    _c=QMetaObject::WriteProperty, _id=2, _a=0x58fd4a0)
    at PlotJuggler/build/3rdparty/qwt/src/qwt_static_autogen/include/moc_qwt_plot_opengl_canvas.cpp:192
#3  0x00007ff81d083917 in QMetaProperty::write(QObject*, QVariant const&) const () from PlotJuggler/install/bin/Qt5Core.dll
#4  0x00007ff81d0abfd9 in QObject::setProperty(char const*, QVariant const&)
    () from PlotJuggler/install/bin/Qt5Core.dll
#5  0x00000000004fe55e in QwtPlotAbstractCanvas::QwtPlotAbstractCanvas (
    this=0xe1d2f30, canvasWidget=0xe1d2f00)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_abstract_canvas.cpp:541
#6  0x0000000000501ff5 in QwtPlotAbstractGLCanvas::QwtPlotAbstractGLCanvas (
    this=0xe1d2f30, canvasWidget=0xe1d2f00)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_abstract_canvas.cpp:897
#7  0x0000000000503cc7 in QwtPlotOpenGLCanvas::QwtPlotOpenGLCanvas (
    this=0xe1d2f00, plot=0x0)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_opengl_canvas.cpp:51
#8  0x0000000000447ef8 in PlotWidget::PlotWidget (this=0xe11ecd0,
    datamap=..., parent=0xe11af60)
    at PlotJuggler/src/plotwidget.cpp:153
#9  0x0000000000465a02 in DockWidget::DockWidget (this=0xe11af60,
    datamap=..., parent=0xe117520)
    at PlotJuggler/src/plot_docker.cpp:271
#10 0x000000000046272b in PlotDocker::PlotDocker(QString, PJ::PlotDataMapRef&, QWidget*)::$_0::operator()() const (this=0x58fde70)
    at PlotJuggler/src/plot_docker.cpp:34
#11 0x0000000000462474 in PlotDocker::PlotDocker (this=0xe117520, name=...,
    datamap=..., parent=0xe113070)
    at PlotJuggler/src/plot_docker.cpp:49
#12 0x000000000046d2ac in TabbedPlotWidget::addTab (this=0xe113070,
    tab_name=...)
    at PlotJuggler/src/tabbedplotwidget.cpp:133
#13 0x000000000046c863 in TabbedPlotWidget::TabbedPlotWidget (this=0xe113070,
    name=..., mainwindow=0x58ff750, mapped_data=..., parent=0x58ff750)
    at PlotJuggler/src/tabbedplotwidget.cpp:83
#14 0x0000000000419ab6 in MainWindow::MainWindow (this=0x58ff750,
    commandline_parser=..., parent=0x0)
    at PlotJuggler/src/mainwindow.cpp:132
#15 0x0000000000417953 in main (argc=1, argv=0x6199a80)
    at PlotJuggler/src/main.cpp:243
(gdb) c
Continuing.

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000502474 in QwtPlotAbstractGLCanvas::setLineWidth (this=0xe127630,
    width=2)
    at PlotJuggler/3rdparty/qwt/src/qwt_plot_abstract_canvas.cpp:1027
1027        if ( width != d_data->lineWidth )
(gdb) c
Continuing.
[Thread 1468.0x7924 exited with code 3221225477]
[Thread 1468.0x8250 exited with code 3221225477]
[Thread 1468.0x4a58 exited with code 3221225477]
[Thread 1468.0x7898 exited with code 3221225477]
[Thread 1468.0x6c8c exited with code 3221225477]
[Thread 1468.0x4ffc exited with code 3221225477]
[Thread 1468.0x1768 exited with code 3221225477]
[Thread 1468.0x561c exited with code 3221225477]
[Thread 1468.0x3678 exited with code 3221225477]
[Thread 1468.0x7d28 exited with code 3221225477]
[Thread 1468.0x64d8 exited with code 3221225477]
[Thread 1468.0x4cf0 exited with code 3221225477]
[Thread 1468.0x7d08 exited with code 3221225477]
[Thread 1468.0x6764 exited with code 3221225477]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
(gdb)

```

I will note that behavior in gdb is different than outside gdb - within gdb it crashes after splash and before the main window opens, but without gdb it crashes when data is dragged onto a plot. The backtrace makes me think they're correlated, but since the .exe behaves differently depending on how it's run I'm not sure.

I'm not familiar enough with Qt and c++ to understand what's going on - any ideas?

jlmcmchl, 2021-01-03:
I figure this out last night - there's something about the QwtPlotOpenGLCanvas causing a crash on windows. Reverting plotwidget.cpp to use `QwtPlotCanvas` from `QwtPlotOpenGLCanvas` fixed it.

Everything except for LSL builds in my environment - for those curious there's multiple compilation errors going on that I assume are related to my setup, but it might be a general thing. 

PR #379 has been submitted to show what I did to get windows running.

I stopped at these errors for LSL, as it seems more likely that it was unable to locate the lslboost folder for headers than anything, which might just be an issue with my setup. The dllimport errors, on the other hand, are pretty opaque to me. 
```make
Scanning dependencies of target lslobj
[ 62%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/lslobj_autogen/mocs_compilation.cpp.obj
[ 63%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/api_config.cpp.obj
[ 63%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/cancellation.cpp.obj
[ 63%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/cast.cpp.obj
[ 64%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/common.cpp.obj
/H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/common.cpp:18:22: error: dllimport cannot be applied to non-inline function definition
LIBLSL_C_API int32_t lsl_protocol_version() {
                     ^
/H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/common.cpp:22:22: error: dllimport cannot be applied to non-inline function definition
LIBLSL_C_API int32_t lsl_library_version() { return LSL_LIBRARY_VERSION; }
                     ^
/H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/common.cpp:24:21: error: dllimport cannot be applied to non-inline function definition
LIBLSL_C_API double lsl_local_clock() {
                    ^
/H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/common.cpp:30:19: error: dllimport cannot be applied to non-inline function definition
LIBLSL_C_API void lsl_destroy_string(char *s) {
                  ^
/H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/common.cpp:34:26: error: dllimport cannot be applied to non-inline function definition
LIBLSL_C_API const char *lsl_last_error(void) { return last_error; }
                         ^
5 errors generated.
make[2]: *** [3rdparty/liblsl/CMakeFiles/lslobj.dir/build.make:134: 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/common.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1092: 3rdparty/liblsl/CMakeFiles/lslobj.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

```make
[ 63%] Building CXX object 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/consumer_queue.cpp.obj
In file included from /H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/consumer_queue.cpp:1:
In file included from /H/Msys2/home/Jordan McMichael/PlotJuggler/./3rdparty/liblsl/src/consumer_queue.h:5:
/H/Msys2/home/Jordan McMichael/PlotJuggler/./3rdparty/liblsl/src/forward.h:28:18: error: no template named 'intrusive_ptr' in namespace 'lslboost'; did you mean 'boost::intrusive_ptr'?
using sample_p = lslboost::intrusive_ptr<class sample>;
                 ^~~~~~~~~~~~~~~~~~~~~~~
                 boost::intrusive_ptr
\H\Msys2\mingw64\include\boost/smart_ptr/intrusive_ptr.hpp:53:25: note: 'boost::intrusive_ptr' declared here
template<class T> class intrusive_ptr
                        ^
In file included from /H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/consumer_queue.cpp:1:
/H/Msys2/home/Jordan McMichael/PlotJuggler/./3rdparty/liblsl/src/consumer_queue.h:17:22: error: no member named 'lockfree' in namespace 'lslboost'; did you mean 'boost::lockfree'?
        using buffer_type = lslboost::lockfree::spsc_queue<sample_p>;
                            ^~~~~~~~~~~~~~~~~~
                            boost::lockfree
\H\Msys2\mingw64\include\boost/lockfree/spsc_queue.hpp:40:11: note: 'boost::lockfree' declared here
namespace lockfree {
          ^
In file included from /H/Msys2/home/Jordan McMichael/PlotJuggler/3rdparty/liblsl/src/consumer_queue.cpp:2:
/H/Msys2/home/Jordan McMichael/PlotJuggler/./3rdparty/liblsl/src/sample.h:24:2: error: no member named 'endian' in namespace 'lslboost'; did you mean 'boost::endian'?
        lslboost::endian::endian_reverse_inplace(t);
        ^~~~~~~~~~~~~~~~
        boost::endian
\H\Msys2\mingw64\include\boost/endian/conversion.hpp:26:11: note: 'boost::endian' declared here
namespace endian
          ^
3 errors generated.
make[2]: *** [3rdparty/liblsl/CMakeFiles/lslobj.dir/build.make:147: 3rdparty/liblsl/CMakeFiles/lslobj.dir/src/consumer_queue.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1092: 3rdparty/liblsl/CMakeFiles/lslobj.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

facontidavide, 2021-01-05:
@csavur  said he would be responsible for the LSL plugin. I don't want to be the maintainer of that plugin, therefore I will just remove it from windows.

I will check again if compilation on Windows works or not and release the new binary installer soon

csavur, 2021-01-05:
> @csavur said he would be responsible for the LSL plugin. I don't want to be the maintainer of that plugin, therefore I will just remove it from windows.
> 
> I will check again if compilation on Windows works or not and release the new binary installer soon

Hello,

I am working on it.

csavur, 2021-01-05:
I made changes to liblsl cmake and send a PR. Can you please accept it?

facontidavide, 2021-01-26:
I just compiled the latest master version in both MSVC and MingW and everything look fine. I will provide soon the windows installer for version 3.1

csavur, 2021-01-27:
Great! I am glad everything was ok.

seanmcleod, 2021-01-30:
@facontidavide any update in terms of the Windows build for version 3.1?

facontidavide, 2021-01-31:
released

closed 2021-01-31 by facontidavide

roger-, 2021-06-01:
The 3.1.1 Windows installer still seems to be missing the lsl plugin?