PlotJuggler/PlotJuggler #370: libDataStreamMQTT plugin does not build

timonegk, 2020-12-21:
On my setup, building the libDataStreamMQTT plugin failes with the following error message:
```
/usr/bin/ld: ../../devel/lib/libpaho-mqtt3a.a(MQTTAsync.c.o): relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
```
It does build if I build paho.mqtt.c as a shared library (switch bools in CMakeLists.txt:219,220 and remove -static suffix in plugins/DataStreamMQTT/CMakeLists.txt:17,18). I don't know though if that's a valid fix or if the library is built as static for a reason.

facontidavide, 2021-01-05:
How can I reproduce this?

timonegk, 2021-01-05:
Sorry; the problem occurs when building with clang (on both Debian Testing and Ubuntu 18.04)

facontidavide, 2021-01-05:
fixed

closed 2021-01-05 by facontidavide