PlotJuggler/PlotJuggler #371: Compilation problem

znlangel, 2020-12-22:
Thanks for contributing to PlotJuggler. You are great!

I downloaded the latest code，But when I compile with Qt 5.14.2，
Compile error：defaulted default constructor cannot be constexpr because the corresponding implicitly declared default constructor would not be constexpr，i check I checked the information on the Internet and found that this is a Qt bug，so I submit my solution：Finally I downloaded Qt 5.9.0, recompiled and passed。
this Is the Documentation on the web：https://bugreports.qt.io/browse/QTBUG-85644

facontidavide, 2020-12-24:
Anything i need to do on my side?

seanmcleod, 2020-12-31:
@znlangel are you compiling using Visual Studio?

According to the bug report that you posted this doesn't result in a compiler error, it's simply an Intellisense error, i.e. the build actually builds and generates an .exe.

facontidavide, 2021-01-05:
I guess I will close this, if there isn't anything I am supposed to do...

closed 2021-01-05 by facontidavide