PlotJuggler/PlotJuggler #372: Couldn't find executable named PlotJuggler below /opt/ros/melodic/share/plotjuggler

whyscience, 2020-12-22:
Help Just after I update my Ubuntu 18.04 to the latest, I can't find executable named PlotJuggler or plotjuggler

```
➜  Downloads rosrun plotjuggler PlotJuggler  

[rosrun] Couldn't find executable named PlotJuggler below /opt/ros/melodic/share/plotjuggler
➜  Downloads rosrun plotjuggler plotjuggler  
[rosrun] Couldn't find executable named plotjuggler below /opt/ros/melodic/share/plotjuggler
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /opt/ros/melodic/share/plotjuggler

```

whyscience, 2020-12-22:
I install use 
```
sudo apt install ros-melodic-plotjuggler  
sudo apt install ros-melodic-plotjuggler-ros
```

facontidavide, 2020-12-22:
The new name of the executable is lower case

whyscience, 2020-12-24:
> The new name of the executable is lower case

Yes, but still no plotjugger found

```
➜  plotjuggler_ros roslaunch plotjuggler_ros plotjuggler.launch

RLException: [plotjuggler.launch] is neither a launch file in package [plotjuggler_ros] nor is [plotjuggler_ros] a launch file name
The traceback for the exception was written to the log file
➜  plotjuggler_ros rosrun plotjuggler plotjugger               
[rosrun] Couldn't find executable named plotjugger below /opt/ros/melodic/share/plotjuggler

```

After I installed, no bin of plotjugger found on latest Ubuntu 18.04

whyscience, 2020-12-24:
There may be something wrong with Ubuntu distro

whyscience, 2020-12-24:
You may just have a try, remove it first and install it later to see whether it is OK

facontidavide, 2020-12-24:
You are right, there is something fishy. I am investigating

facontidavide, 2020-12-24:
fixed. I need to urgently release 3.0.6 with this fix. Thanks for reporting

closed 2020-12-24 by facontidavide

facontidavide, 2020-12-24:
Important. If you are afected by this problem, use the **snap** version until 3.0.6 is made available in your ROS distribution

whyscience, 2020-12-26:
> snap

Thanks

Kin-Zhang, 2020-12-30:
But the problem didn't solve if you install plotjuggler in this way
`sudo apt install ros-melodic-plotjuggler-ros`
It still:Couldn't find executable named PlotJuggler below /opt/ros/melodic/share/plotjuggler
Or RLException: [plotjuggler] is neither a launch file in package [plotjuggler_ros] nor is [plotjuggler_ros] a launch file name
The traceback for the exception was written to the log file

facontidavide, 2020-12-30:
Problem solved in version 3.0.6, that is not released yet as a Debian package.

Releases are announced here https://discourse.ros.org/c/release/melodic/38

Sorry for the inconvenience

Kin-Zhang, 2020-12-30:
Thanks for the replay! I really love plotjuggler too.
The `snap install plotjuggler` is really slow.... I will wait for debian package ready on ROS since we have mirror links on ROS
Thanks again.

Kin-Zhang, 2020-12-30:
Thanks for replying! Love PlotJuggler!
`sudo snap install plotjuggler`

prcgnn, 2022-07-15:
still I have this issue

facontidavide, 2022-07-15:
@prcgnn how do you install it and how are you trying to run it?

facontidavide, 2022-07-15:
there is no roslaunch, only rosrun