PlotJuggler/PlotJuggler #374: ROS subscriber, problem with varying size arrays.

ozzdemir, 2020-12-24:
When visualizing elements of arrays with varying size, pj cannot receive values from the topic that publishes as an array with varying size. I added a video that shows the problem and a bag file to reproduce it.
https://user-images.githubusercontent.com/30624975/103084160-6772a900-45ef-11eb-96e2-246be5888346.mp4

[pj-test-varying-size-problem_bag.zip](https://github.com/facontidavide/PlotJuggler/files/5739334/pj-test-varying-size-problem_bag.zip)

facontidavide, 2020-12-24:
You forgot this variable. 

[image]

closed 2020-12-24 by facontidavide

ozzdemir, 2020-12-24:
It was so obvious :sweat_smile:, thanks for reminding.