PlotJuggler/PlotJuggler #377: Enhancement: Add streaming log output

flybotix, 2020-12-28:
I appreciate what this tool does, even before I've started using it! I'd like to use it for my robotics team's FRC robot.

Those robots (as well as other production time-series data-based systems such as sonar/radar) also include streaming timestamped text data in the form of logs, across OS/network boundaries. The next 'big thing' for this app could be to add a cursor-linked panel to this program that shows the relevant logs surrounding the timestamp of the cursor. I have some high-level ideas on how to keep this feature from getting too complex in its first iteration, given my current understanding of how PlotJuggler works on the front-end.

Looking forward to showing this to my students in the coming weeks!

facontidavide, 2020-12-28:
There is a SPECIFIC class of plugins called "Publishers" to do this, therefore it is pretty easy!!!!

https://github.com/facontidavide/PlotJuggler/blob/main/include/PlotJuggler/statepublisher_base.h

facontidavide, 2020-12-28:
Since there is no action required from my side, I will close this, but feel free to open it again if necessary

closed 2020-12-28 by facontidavide

flybotix, 2020-12-28:
@facontidavide you seem to have pointed me to a way to extend the core functionality via writing C++ code, but I don't do C++. Is there a way to do this out of the box with just a bit of configuration, e.g. just loading a CSV file and a corresponding log file?

facontidavide, 2020-12-28:
I am confused (or you are).

You may connect plotjuggler to your robot as long as you use one of the available protocols and transport mechanism. 

For instance, you may read data sent through UDP and formatted in JSON. 

Or you may load data from a CSV file. 

But you mentioned "cursor-linked panel to this program that shows the relevant logs surrounding the timestamp of the cursor". If I understand what you mean, this is already supported in the left panel.

Move the cursor and you will see the value of each time series related to that timestamp.

flybotix, 2020-12-28:
Sorry for the confusion!

The "logs" I'd like to see are the terminal outputs of the robot code.

For example, say I have a CSV file that contains the various numerical-based sensor data of a robot, and another file (that can be formatted in CSV, JSON, etc) that contains timestamps with terminal output lines (e.g. `time=0.013, log=com.flybotix [DEBUG] "The robot has initiated autonomous mode"` ). As I move the mouse around a graph in one panel, in a different panel I would like to see the corresponding debug lines around the timestamp of the cursor on the graph.

facontidavide, 2020-12-28:
This functionality is implemented in ROS. If you are not using ROS, you need to implement it yourself with a c++ plugin.

It is not too much work... if you know c++. You may look at the equivalent plugin in ROS to see how that works.

https://github.com/PlotJuggler/plotjuggler-ros-plugins/tree/development/plugins/RosoutPublisher