PlotJuggler/PlotJuggler #380: Compilation problem on conda-forge

Tobias-Fischer, 2021-01-05:
Hi there,

We are maintaining https://github.com/RoboStack/ros-noetic which brings ROS and conda together. We used to compile plotjuggler successfully on Linux, Win and OSX. Recent changes broke our builds.

I'd like to tackle Linux first. There we get a lot of errors related to liblsl: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1184&view=logs&j=2edfdb19-aeb9-584c-1cd6-ebd6d631236d&t=afd38b15-9dba-5ca8-f177-685dc0e02b80

Any help in getting plotjuggler to compile again would be very appreciated :).

Best, Tobias

facontidavide, 2021-01-05:
@csavur this is failing because of LSL. Please provide a solution, since you are the maintainer of that plugin.

csavur, 2021-01-05:
The problem happening because of CMake. I am working on it.  @Tobias-Fischer can you guide me on how can I compile plotjuggler against your pipeline?

Tobias-Fischer, 2021-01-05:
Hi @facontidavide @csavur, many thanks for your quick responses!

The guidelines are available here: https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md

In the vinca.yaml, under `packages_select_by_deps:` just leave in `- plotjuggler` and `-plotjuggler_ros` and comment out all other packages.

Tobias-Fischer, 2021-01-05:
For OSX, the errors can be seen here: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1209&view=logs&j=289a654c-19bd-520a-954b-f92d58d426d5&t=4092ab7a-b52d-5c57-9638-c345cacc748f

There are some odd redefinitions going on:
```
In file included from plugins/DataStreamROS_autogen/mocs_compilation.cpp:2:
In file included from plugins/DataStreamROS_autogen/L2RW7QHEBF/moc_datastream_ROS.cpp:9:
In file included from plugins/DataStreamROS_autogen/L2RW7QHEBF/../../../../ros-noetic-plotjuggler-ros/src/work/plugins/DataStreamROS/datastream_ROS.h:14:
In file included from $SRC_DIR/ros-noetic-plotjuggler-ros/src/work/plugins/ros1_parsers/ros1_parser.h:5:
$PREFIX/include/PlotJuggler/messageparser_base.h:27:7: error: redefinition of 'MessageRef'
class MessageRef
      ^
$PREFIX/include/PlotJuggler/PlotJuggler/messageparser_base.h:27:7: note: previous definition is here
class MessageRef
      ^
In file included from plugins/DataStreamROS_autogen/mocs_compilation.cpp:2:
In file included from plugins/DataStreamROS_autogen/L2RW7QHEBF/moc_datastream_ROS.cpp:9:
In file included from plugins/DataStreamROS_autogen/L2RW7QHEBF/../../../../ros-noetic-plotjuggler-ros/src/work/plugins/DataStreamROS/datastream_ROS.h:14:
In file included from $SRC_DIR/ros-noetic-plotjuggler-ros/src/work/plugins/ros1_parsers/ros1_parser.h:5:
```

csavur, 2021-01-05:
> Hi @facontidavide @csavur, many thanks for your quick responses!
>
> The guidelines are available here: https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md
>
> In the vinca.yaml, under [...]

@Tobias-Fischer Where can I change the URL for the plot juggles to my own repo to test changes? I could not find it.

Tobias-Fischer, 2021-01-05:
Unfortunately this is not possible at the moment. Instead, you have to use a patch - see for example https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-plotjuggler.win.patch (`.win.patch` indicates the patch is only applied on Windows - instead use `.patch` if you want to apply the patch on all platforms)

csavur, 2021-01-05:
> Unfortunately this is not possible at the moment. Instead, you have to use a patch - see for example https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-plotjuggler.win.patch [...]

I see this is one way which is difficult.

An alternative solution, I guess you are cloning the source code to somewhere? What if I made changes there and tested quickly to see the result. Where do you clone source code? I hope each time I called the "boa build ." it does not clone fresh.

Tobias-Fischer, 2021-01-05:
Ahh yes. This is possible indeed. @wolfv might be able to give you instructions on how to do this nicely :)

Tobias-Fischer, 2021-01-05:
Actually, in the recipe.yaml that is created, you can easily change the git_url and git_rev.

csavur, 2021-01-05:
Can you give me an example? How to add git_url?

csavur, 2021-01-05:
I found it. Thank you.

csavur, 2021-01-05:
@Tobias-Fischer  

git_url and git_rev did not work since it is cloning specific version. Instead, I am trying to use the patch however It does not apply the patch. Do I need to change anything to patch to be applied before compilation?

Tobias-Fischer, 2021-01-06:
You need to re-run vinca so that a recipe.yaml is generated that includes the patch. Or alternatively add the patch manually, e.g. (here for a different package but you'll get the idea)
```
package:
  name: ros-noetic-moveit-ros-perception
  version: 1.1.0
source:
  git_url: https://github.com/ros-gbp/moveit-release.git
  git_rev: release/noetic/moveit_ros_perception/1.1.0-1
  folder: ros-noetic-moveit-ros-perception/src/work
  patches:
    - patch/ros-noetic-moveit-ros-perception.win.patch
```

csavur, 2021-01-06:
> You need to re-run vinca so that a recipe.yaml is generated that includes the patch. Or alternatively add the patch manually, e.g. (here for a different package but you'll get the idea)
>
> ``` [...]

@Tobias-Fischer  Thank you. It is taking time to debug the code. For now, I disabled the new plugin we added (LSL) to plotjuggler.

You can access the patch from [https://gist.github.com/csavur/7d86fec19568f6177283300483534efa]

I will fix this problem when I have more free time. I am a little busy this week, hopefully, I will take a look next week. I will post here as soon as I fixed it.

Best,

Tobias-Fischer, 2021-01-06:
Great, I can now compile PlotJuggler again without issues on all platforms. Thanks!

closed 2021-01-23 by facontidavide