PlotJuggler/PlotJuggler #382: Improve exception safety with smart pointers

elfring, 2021-01-11:
Would you like to wrap any pointer data members with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

Update candidates:
- [MovingAverageFilter](https://github.com/facontidavide/PlotJuggler/blob/e7ac8cfbc2a543d243e5e5a1497444fc5e07b7ac/src/transforms/moving_average_filter.cpp#L6)
- [OutlierRemovalFilter](https://github.com/facontidavide/PlotJuggler/blob/e7ac8cfbc2a543d243e5e5a1497444fc5e07b7ac/src/transforms/outlier_removal.cpp#L4)

facontidavide, 2021-01-11:
I am not sure I understand what you propose

elfring, 2021-01-11:
I suggest to [reconsider the usage of C++ new and delete operators](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r20-use-unique_ptr-or-shared_ptr-to-represent-ownership "C++ Guideline “R.20: Use unique_ptr or shared_ptr to represent ownership”").

facontidavide, 2021-01-11:
I know exactly what smart pointers are :stuck_out_tongue: 

My question is why do you feel the need here to use them. I am just following the Qt idiom. 

**In this particular case**, it might make sense, but please remember that Qt has its own way to automatically delete the children of a parent widget. For that reason, it is usually not a great idea to use smartpointers with widgets.

But, as I said, here it might be equivalent.

elfring, 2021-01-11:
>I know exactly what smart pointers are

Why do you use some explicit delete statements (in destructors for example) then?

>For that reason, it is usually not a great idea to use smartpointers with widgets.

Would you like to use the Qt approach for desired resource clean-up at more places?

facontidavide, 2021-01-11:
@are you familiar with Qt development? From your questions it looks like you aren't.

I don't have much bandwidth to spend in bikeshedding. If you want, you may send a Pull Request of well tested changes

closed 2021-01-11 by facontidavide

elfring, 2021-01-11:
:crystal_ball: I became curious under which circumstances this software can become exception-safe in the C++ sense.