PlotJuggler/PlotJuggler #383: Compilation Error: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object

The-personified-devil, 2021-01-11:
I tried compiling the latest version (and the latest tag) of this tool on Fedora 32 and got the following linker error:
`/usr/bin/ld: ../../lib/libpaho-mqtt3a.a(MQTTAsync.c.o): relocation R_X86_64_32 against ".rodata" can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[2]: *** [plugins/DataStreamMQTT/CMakeFiles/DataStreamMQTT.dir/build.make:129: bin/libDataStreamMQTT.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1114: plugins/DataStreamMQTT/CMakeFiles/DataStreamMQTT.dir/all] Error 2
make: *** [Makefile:172: all] Error 2
`
Here is the output of `cmake ..`:

CMake Deprecation Warning at CMakeLists.txt:3 (cmake_policy):
  The OLD behavior for policy CMP0048 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Check for working C compiler: /usr/lib64/ccache/cc
-- Check for working C compiler: /usr/lib64/ccache/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- --------------------------------------------------
-- PlotJuggler is being WITHOUT any ROS support
-- --------------------------------------------------
-- No build type selected, default to RelWithDebInfo
-- Could NOT find ZMQ (missing: ZMQ_LIBRARIES ZMQ_INCLUDE_DIRS) 
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR) 
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR) 
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR) 
-- Found Backward: /home/destroyer/Documents/PlotJuggler/3rdparty/backward-cpp  
-- Submodule update
-- CMake version: 3.17.4
-- CMake system name: Linux
-- Timestamp is 2021-01-11T18:03:21Z
-- Using bundled pugixml
-- Found Git: /usr/bin/git (found version "2.26.2") 
-- Git version information: HEAD/v1.14.0-26-g65ffb693
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for clock_gettime
-- Looking for clock_gettime - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/destroyer/Documents/PlotJuggler/build

facontidavide, 2021-02-28:
sorry, I don't actively support Fedora.

closed 2021-02-28 by facontidavide

martonmiklos, 2021-12-13:
For reference I am getting the same error while building in QtCreator on Ubuntu 20.04 with Qt 5.15

```
/usr/bin/ld: ../../lib/libpaho-mqtt3a.a(MQTTAsync.c.o): relocation R_X86_64_32S against `.data' can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[2]: *** [plugins/DataStreamMQTT/CMakeFiles/DataStreamMQTT.dir/build.make:109: bin/libDataStreamMQTT.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1090: plugins/DataStreamMQTT/CMakeFiles/DataStreamMQTT.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

facontidavide, 2021-12-14:
The Paho based plugin has been replaced with another one based on Mosquitto. I will not support the former anymore

martonmiklos, 2021-12-14:
> The Paho based plugin has been replaced with another one based on Mosquitto. I will not support the former anymore

Argggwwwwwhhhh. I have rebased my work onto the master not the main.
:face_exhaling:

facontidavide, 2021-12-14:
blame woke culture for that ;)

martonmiklos, 2021-12-14:
> blame woke culture for that ;)

I do every time when I create a repo and change the default branch to master manually...
Old habits die hard.