PlotJuggler/PlotJuggler #384: Grouping variables using "/" fails with websockets and JSON

JimmyDaSilva, 2021-01-11:
Hi @facontidavide, 

I am switching to your new plugin for Websockets communication with JSON.
Works great, except the key names via JSON seem to do not like the "/".
"/" is converted to "~1", so I can't group the variables.

To easily reproduce this, I have slightly modified [websocket_client.py](https://github.com/facontidavide/PlotJuggler/blob/main/plugins/DataStreamWebsocket/websocket_client.py
) by adding a variable named: `var/x`

```
#!/usr/bin/python

# https://pypi.org/project/websocket_client/

import websocket
import math
import json
from time import sleep

ws = websocket.WebSocket()
ws.connect("ws://localhost:9871")

time = 0.0

while True:
    sleep(0.05)
    time += 0.05

    data = {
        "timestamp": time,
        "test_data": {
            "cos": math.cos(time),
            "sin": math.sin(time)
        },
        "var/x": 2
    }
    ws.send( json.dumps(data) )
```
Here is the result:
[image]

Is this a bug, or is this a constraint coming from a string encoding?

Grouping variables via JSON object encapsulation works.
Reading a CSV with a slash in the names works too.

I am using the lastest tag 3.0.7

Thanks for version 3, it is amazing!
Jimmy

JimmyDaSilva, 2021-01-13:
To remove one possibility regarding the issue, I made a new function in [websocket_server.cpp](https://github.com/facontidavide/PlotJuggler/blob/main/plugins/DataStreamWebsocket/websocket_server.cpp) connected to the WebSocket::binaryMessageReceived signals, and sent directly binary messages (QByteArray) from my c++ Client.
The result is still the same as above.

So I guess the problem comes from the `reinterpret_cast` here: https://github.com/facontidavide/PlotJuggler/blob/main/plugins/DataStreamWebsocket/websocket_server.cpp#L185

JimmyDaSilva, 2021-01-13:
I checked the cast. It is okay.

Printing data at every step got me to this line:
https://github.com/facontidavide/PlotJuggler/blob/main/src/nlohmann_parsers.cpp#L50

The conversion from `/` to `~1` happens here, unfortunately.
So I guess we are stuck. Meaning grouping variables when using JSON cannot be done with a forward slash.

facontidavide, 2021-01-14:
well, what I can do is to reimplement that particular function. I don't expect it to be a big deal

JimmyDaSilva, 2021-01-14:
Maybe using the integrated JSON functions of Qt?
https://stackoverflow.com/questions/25977752/how-to-format-a-json-string-to-a-readable-output-in-qt

facontidavide, 2021-01-14:
fixed!

closed 2021-01-14 by facontidavide

JimmyDaSilva, 2021-01-14:
Wow, that was fast !
Thanks Davide :)