PlotJuggler/PlotJuggler #385: Allow custom lua functions to return an arbitrary number of samples

Hugal31, 2021-01-14:
For now, Lua functions must return exactly one point. It would be great to be able to return any amount of point (including 0).

My use-case is this-one: I want to reproduce the [step](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.step.html) behavior of pyplot, and for this I could create a custom function which would return a second sample just before the new one, that is equal to the previous value.

facontidavide, 2021-01-14:
This is a VERY specific use case :( 

I can see the option of returning zero point, but if you return more than a single point, then they will share the same timestamp, that is ok for YOUR use case, but nothing more.

So... nope!

Hugal31, 2021-01-14:
It is already possible to give a timestamp with custom functions, so I do not see why the two samples will share the same timestamp.

Example:
```lua
-- Valid function today
return time + 0.4, value * 2
-- My proposition
return {{time + 0.4, value * 2}, {time + 0.5, value * 3}}
```

facontidavide, 2021-01-14:
this sound more reasonable :) 

It can be definitively done. I will keep the issue open as a reminder, but it is not in my current TODO list.

PR #386 (2021-01-23): Allow custom function return multiple points
Remplace `CustomFunction::calculatePoint` by `CustomFunction::calculatePoints`, which takes a caller-allocated vector reference in parameter (instead of returning one, to reduce the number of allocations), and can therefore insert multiple points.

Lua functions can now take advantage of this by returning an array of two-number arrays:

```lua
-- Still valid
return value
return time, value
-- New syntax
return {{time - 0.01, value * 0.1}, {time, value * 1.1}}
```

BREAKING CHANGE: `CustomFunction::calculatePoint` is replaced by `CustomFunction::calc [... cut]
closed 2021-01-23 by facontidavide