PlotJuggler/PlotJuggler #387: Core dump in PlotJuggler-3 3.0.7

clydemcqueen, 2021-01-15:
Thanks for contributing to PlotJuggler. You are great!

# If you ignore these instructions, I will ignore your issue

Please allow me to better understand your problem providing the following information:

- Please use always the master branch or the latest tag, if you can. 

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

[image]

Yay, thanks for PlotJuggler!
I'm using the latest debians on Ubuntu 20.04:
PlotJuggler-3 3.0.7

When reading from the attached bag I get a core dump:

```
clyde@fastr:~/ros2/orca3_ws$ /opt/ros/foxy/lib/plotjuggler/plotjuggler
Loading compatible plugins from directory:  "/opt/ros/foxy/lib/plotjuggler_ros"
"libDataLoadROS2.so" : is a DataLoader plugin
"libDataStreamROS2.so" : is a DataStreamer plugin
[WARN] rclcpp: logging was initialized more than once
"libTopicPublisherROS2.so" : is a StatePublisher plugin
Number of plugins loaded:  3 

Loading compatible plugins from directory:  "/opt/ros/foxy/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamLSL.so" : is a DataStreamer plugin
"libDataStreamMQTT.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
Number of plugins loaded:  8 

Loading compatible plugins from directory:  "/home/clyde/.local/share/PlotJuggler"
Number of plugins loaded:  0 

[INFO] rosbag2_storage: Opened database '/home/clyde/ros2/orca3_ws/bad_odom/bad_odom.bag_0.db3' for READ_ONLY.
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f35fe1fd4da, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f35fe256564, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f35fd1b22e2, in g_main_context_iteration
#28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f35fd1b223f, in g_main_context_dispatch
#27   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f35fd1b1fbc, in g_main_context_dispatch
#26   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f35fa41632d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#25   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f35fe5c335a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#24   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f35fe5e910a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#23   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f35fe5e77d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f35fe1fe939, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec270ef, in QApplication::notify(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec1da65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec7f1eb, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec7c35c, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec26456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#16   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f35fe1fe939, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#15   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec27342, in QApplication::notify(QObject*, QEvent*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec1da65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fec602b5, in QWidget::event(QEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fed14034, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fed13e72, in QAbstractButton::setIconSize(QSize const&)
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fed12a2d, in QAbstractButton::toggled(bool)
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f35fed12805, in QAbstractButton::clicked(bool)
#8    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f35fe22a3bb, in QMetaObject::activate(QObject*, int, int, void**)
#7    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x56120739e7b9, in 
#6    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x56120739d45d, in 
#5    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x5612073ce490, in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag)
#4    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x5612073c75b6, in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag)
#3    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x5612073c3f3a, in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag)
#2    Object "/opt/ros/foxy/lib/plotjuggler_ros/libDataLoadROS2.so", at 0x7f35f0430fab, in DataLoadROS2::readDataFromFile(PJ::FileLoadInfo*, PJ::PlotDataMapRef&)
#1    Object "/opt/ros/foxy/lib/plotjuggler_ros/libDataLoadROS2.so", at 0x7f35f0438c87, in CompositeParser::parseMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_uint8_array_t const*, double)
#0    Object "/opt/ros/foxy/lib/plotjuggler_ros/libDataLoadROS2.so", at 0x7f35f0445273, in BuiltinMessageParser<nav_msgs::msg::Odometry_<std::allocator<void> > >::parseMessage(rcutils_uint8_array_t const*, double)
Segmentation fault (Address not mapped to object [(nil)])
Segmentation fault (core dumped)

```

I haven't yet tried to build from source, but that is my plan for later today.

[bad_odom.gz](https://github.com/facontidavide/PlotJuggler/files/5822087/bad_odom.gz)

Thanks,
/Clyde

facontidavide, 2021-01-15:
I am not sure I got the file right. If this is a rosbag2, shouldn't it be a folder?

clydemcqueen, 2021-01-15:
[bad_odom.tar.gz](https://github.com/facontidavide/PlotJuggler/files/5822746/bad_odom.tar.gz)

facontidavide, 2021-01-20:
Problem solved in the other repo:

https://github.com/PlotJuggler/plotjuggler-ros-plugins/commit/632a2a051697b5f0fa08dad2e8c601d12af880b7

Thanks for reporting

closed 2021-01-20 by facontidavide