PlotJuggler/PlotJuggler #388: Time/Distance measurement

g3n35i5, 2021-01-18:
Hi again! So far, i love the 3.X version of PlotJuggler, great work!

I have a small feature request (If its already implemented and i cant find it: please blame me publicly :smile:)

It would be great if you could measure distances in the X/Y plane (delta of time, delta of curve, ...). I would imagine a kind of "ruler" as a implementation.

[image]

facontidavide, 2021-01-18:
No, unfortunately there isn't such a feature. It seems very specific and I doubt that I will work on this any time soon, sorry.

g3n35i5, 2021-01-18:
All right, it was worth a try :D Thanks for the quick reply!

closed 2021-01-20 by facontidavide