PlotJuggler/PlotJuggler #390: Color/Palette issue on Windows

JimmyDaSilva, 2021-01-19:
Hi Davide,

I know there is no Windows release yet, but I wanted to use it now anyway...
I built the last version(https://github.com/facontidavide/PlotJuggler/commit/6dc10a385591bc097b2b69e4367cbfa3ffe06d59) of PlotJuggler from source with QT 5.15.2 and MSVC 2017. 
I got it to compile and run.

But there is a weird color issue. Can't say where it comes from.
Light theme:
[image]

Dark theme:
[image]

facontidavide, 2021-01-19:
well, that looks really bad :( 

I need to have a look at this

facontidavide, 2021-01-24:
fixed. Thanks a lot!

closed 2021-01-24 by facontidavide