PlotJuggler/PlotJuggler #391: Is it working for new M1 macs, I am trying for more than 2 hours now.

Shiv3901, 2021-01-21:
It crashed after 72% build.

[ 71%] Built target paho-mqtt3a-static
[ 72%] Automatic MOC and UIC for target DataStreamMQTT
[ 72%] Built target DataStreamMQTT_autogen
[ 72%] Linking CXX shared library ../../bin/libDataStreamMQTT.dylib
ld: warning: ignoring file /usr/local/opt/qt5/lib/QtWidgets.framework/QtWidgets, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/opt/qt5/lib/QtCore.framework/QtCore, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/opt/qt5/lib/QtGui.framework/QtGui, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "qrand()", referenced from:
      MQTT_Dialog::MQTT_Dialog() in datastream_mqtt.cpp.o
  "QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)", referenced from:
      ConnectionLost(void*, char*) in datastream_mqtt.cpp.o
      QDebug::operator<<(char const*) in datastream_mqtt.cpp.o
      QString::~QString() in datastream_mqtt.cpp.o
      onSubscribeFailure(void*, MQTTAsync_failureData*) in datastream_mqtt.cpp.o
      onConnectFailure(void*, MQTTAsync_failureData*) in datastream_mqtt.cpp.o
      onConnect(void*, MQTTAsync_successData*) in datastream_mqtt.cpp.o
      QString::toStdString() const in datastream_mqtt.cpp.o
      ...
  "QArrayData::shared_null", referenced from:
      DataStreamMQTT::DataStreamMQTT() in datastream_mqtt.cpp.o
      Ui_DataStreamMQTT::setupUi(QDialog*) in datastream_mqtt.cpp.o
  "QBoxLayout::addLayout(QLayout*, int)", referenced from:
      Ui_DataStreamMQTT::setupUi(QDialog*) in datastream_mqtt.cpp.o
  "QBoxLayout::addWidget(QWidget*, int, QFlags<Qt::AlignmentFlag>)", referenced from:
      DataStreamMQTT::start(QStringList*) in datastream_mqtt.cpp.o
      Ui_DataStreamMQTT::setupUi(QDialog*) in datastream_mqtt.cpp.o
  "QFormLayout::setLayout(int, QFormLayout::ItemRole, QLayout*)", referenced from:
      Ui_DataStreamMQTT::setupUi(QDialog*) in datastream_mqtt.cpp.o
  "QFormLayout::setWidget(int, QFormLayout::ItemRole, QWidget*)", referenced from:
      Ui_DataStre

devnexen, 2021-01-23:
hint `building for macOS-arm64 but attempting to link with file built for macOS-x86_64` you re trying to build a native executable with x86 libraries.

facontidavide, 2021-01-23:
Sorry, but I am not supporting macOS officially. If anybody has a solution, he/she is welcome to submit it.

closed 2021-01-24 by facontidavide

pgorczak, 2021-02-23:
I made PlotJuggler work on an M1 Mac so sharing my experience here :)

* Install command line developer tools (Xcode not needed).
* Get Qt5 [source](https://www.qt.io/offline-installers) - I got the 5.12.x source package because I was linked to it from somewhere.
* For compilation, I used the [build from Git](https://wiki.qt.io/Building_Qt_5_from_Git) guide as a base but it boils down to just running the `configure`script with options `-developer-build -opensource -nomake examples -nomake tests -device-option QMAKE_APPLE_DEVICE_ARCHS=arm64` and then `make -j8`. I did none of the other steps in the guide. The additional *arm64* arg comes from the [Qt issue tracker](https://bugreports.qt.io/browse/QTBUG-85279).
* Install CMake, download PlotJuggler, open PlotJuggler in CMake
* Add two entries in CMake:
    * Qt5_DIR = [wherever you built Qt5 to]/qtbase/lib/cmake/Qt5
    * CMAKE_OSX_SYSROOT = /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk
* Configure with Unix Makefiles and generate, switch to CMake build dir and *make -j8*
* Run plotjuggler from the bin directory (I got a segfault on the first run, then opened the binary from finder, and it worked ever since)

mengchaoheng, 2023-03-07:
@pgorczak can you help me? about this [issue](https://github.com/facontidavide/PlotJuggler/pull/725#issuecomment-1438769494)