PlotJuggler/PlotJuggler #398: roslaunch plotjuggler_ros plotjuggler.launch returns an error

Acwok, 2021-02-03:
Hello @facontidavide ,

I can't launch plotjuggler anymore on ros melodic Ubuntu 20.04.
I installed it using: `sudo apt install ros-melodic-plotjuggler`, and when I launch it using: `roslaunch plotjuggler_ros plotjuggler.launch` I get:
```
RLException: [plotjuggler.launch] is neither a launch file in package [plotjuggler_ros] nor is [plotjuggler_ros] a launch file name
The traceback for the exception was written to the log file
```

facontidavide, 2021-02-03:
you need to install

        sudo apt install ros-melodic-plotjuggler-ros

facontidavide, 2021-02-03:
and, by the way, you will not have melodic in Ubuntu 20.04, right?

sam-n-johnston, 2021-02-04:
I'm having a similar issue, but with noetic & ubuntu:20.04. I did `apt install -y ros-noetic-plotjuggler-ros`, then I verified the versions:

```
$ apt-cache showpkg ros-noetic-plotjuggler-ros
Package: ros-noetic-plotjuggler-ros
Versions:
1.0.0-1focal.20210112.083945
...
$ apt-cache showpkg ros-noetic-plotjuggler
Package: ros-noetic-plotjuggler
Versions:
3.0.7-1focal.20210107.101906
...
```

So I would assume that [this](https://github.com/facontidavide/PlotJuggler/issues/372#issuecomment-750878449) comment doesn't apply since the version is 3.0.7. But then I get:

```
$ roslaunch plotjuggler_ros plotjuggler.launch
RLException: [plotjuggler.launch] is neither a launch file in package [plotjuggler_ros] nor is [plotjuggler_ros] a launch file name
The traceback for the exception was written to the log file
```

For now I'm gonna install it from source I guess. Let me know if you need more logs/info.

Acwok, 2021-02-04:
> you need to install
> 
> ```
>     sudo apt install ros-melodic-plotjuggler-ros
> ```
Sorry, that's what I meant.

> and, by the way, you will not have melodic in Ubuntu 20.04, right?

Indeed, it was Ubuntu 18.04.

cshastry, 2021-02-04:
Hello @facontidavide ,

I can confirm the issue. Trying to use launch file(on ros noetic) to launch does not work. I looked under /opt/ros/noetic/share/plotjuggler_ros/ and there is no launch file.

ROSRider, 2021-02-06:
Hello. I am on noetic, ubuntu 20.04, and I installed all the plot juggler packages, and there are no launch files.

is there a work around?

cshastry, 2021-02-06:
> Hello. I am on noetic, ubuntu 20.04, and I installed all the plot juggler packages, and there are no launch files.
> 
> is there a work around?

rosrun plotjuggler plotjuggler

facontidavide, 2021-02-08:
I added a fix that will be available in the next release.

In the meantime, use, for ROS1:

      rosrun plotjuggler plotjuggler

and for ROS2:

       ros2 run plotjuggler plotjuggler

closed 2021-02-08 by facontidavide