PlotJuggler/PlotJuggler #399: Cannot start PlotJuggler under ROS2.

whatis777, 2021-02-03:
### Use Case
I'd like to start PlotJuggler with ROS2 - unfortunately I cannot get it running nor can I find related informations on the internet for the issue here

### Environment
* common [ROS2 Foxy installation](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Debians/ ) on Ubuntu Desktop 20.04.1 LTS
* installation of PlotJuggler for ROS2 following this [github tutorial](https://github.com/facontidavide/PlotJuggler#note-for-ros-users)

### The Issue
The [github tutorial](https://github.com/facontidavide/PlotJuggler#note-for-ros-users) contains ROS1 instructions, so I cannot use 
`roslaunch plotjuggler_ros plotjuggler.launch` 
For ROS2 it would be
`ros2 launch plotjuggler_ros plotjuggler.launch`, but then the shell says:

`file 'plotjuggler.launch' was not found in the share directory of package 'plotjuggler_ros' which is at '/opt/ros/foxy/share/plotjuggler_ros'`

### Question

Could you please provide more detailed information in the README how to start plotjuggler_ros for ROS2?

Thanks in advance :-)

cshastry, 2021-02-04:
I think this is related to issue #398

whatis777, 2021-02-07:
A question to the community: has anyone experience using PlotJuggler with ROS2? Does it work in general?

facontidavide, 2021-02-08:
It should work well in Foxy. I am not maintaining older ROS2 versions

facontidavide, 2021-02-08:
I added a fix that will be available in the next release.

In the meantime, use, for ROS1:

      rosrun plotjuggler plotjuggler

and for ROS2:

       ros2 run plotjuggler plotjuggler

closed 2021-02-08 by facontidavide