PlotJuggler/PlotJuggler #402: Auto-reload custom timeseries when changing new file

ndepal, 2021-02-11:
In my workflow I load new log files (usually `.ulg`) frequently, going back and forth between logs.

I have set up a bunch of custom time series that I want to look at for each log. Currently, the custom time series are not updated with the new input data automatically.

When I open a new log file, I have to say "No" to the pop up asking me if I want to delete the custom time series, then click on each individual time series on the bottom left, click on the edit button, and click "Modify Timeseries" without making any changes. This is tedious if there are many custom timeseries and/or I need to switch between log files regularly.

It would be great if each custom timeseries was re-computed automatically when the user answers "No" to deleting them when opening a new file.

lukegluke, 2021-02-27:
As a workaround for this case I use layouts xml files, that I duplicate and manually change filename for the previouslyLoaded_Datafiles and then load them in PlotJuggler.

But there is another not very good thing/bug - on layout saving customMathEquations list get saved in different order without respecting self dependencies in custom timeseries. For example functions `heading` and `yaw` makes deg from rad and then there is another function `yaw-heading` that use result of both previous timeseries. On layout saving PlotJuggler saves `yaw-heading` function first in customMathEquations section and only then `yaw` and `heading`. After loading this file again PlotJuggler complains "Invalid channel name" because it doesn't now what is $$yaw$$ and $$heading$$.

So I would not recommend to use self dependencies custom timeseries in such way, better copy paste code and use self-sufficient functions.

facontidavide, 2026-09-20:
PlotJuggler 4 recomputes derived/custom series automatically when a file is reloaded or replaced (SessionManager::replaceDataset -> DataProcessorService::recomputeForReplacedSources, including chained filters), so the manual 'Modify Timeseries' loop is gone. Closing as addressed in PJ4.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide