PlotJuggler/PlotJuggler #404: Custom plugin not working with snap

iejMac, 2021-02-12:
Building the plugin from source works fine, however the snap version doesn't work:

plotjuggler --plugin_folders path/to/directory/containing/libDataLoadrlog.so
[libDataStreamrlog.so.tar.gz](https://github.com/facontidavide/PlotJuggler/files/5969697/libDataStreamrlog.so.tar.gz)

```
Qt: Session management error: Could not open network socket
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.447\" (uid=1000 pid=45274 comm=\"/snap/plotjuggler/11/opt/ros/melodic/lib/plotjuggl\" label=\"snap.plotjuggler.plotjuggler (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=875 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.447\" (uid=1000 pid=45274 comm=\"/snap/plotjuggler/11/opt/ros/melodic/lib/plotjuggl\" label=\"snap.plotjuggler.plotjuggler (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=875 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
"Object path cannot be empty"
Loading compatible plugins from directory:  "/home/batman/PlotJuggler/build/bin/"
**"
libDataLoadCSV.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataLoadCSV.so'"
"libDataLoadrlog.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataLoadrlog.so'"
"libDataLoadULog.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataLoadULog.so'"
"libDataStreamLSL.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamLSL.so'"
"libDataStreamMQTT.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamMQTT.so'"
"libDataStreamSample.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamSample.so'"
"libDataStreamUDP.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamUDP.so'"
"libDataStreamWebSocket.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamWebSocket.so'"
"libDataStreamZMQ.so" :  "Failed to extract plugin meta data from '/home/batman/PlotJuggler/build/bin/libDataStreamZMQ.so'"**
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/snap/plotjuggler/11/opt/ros/melodic/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamLSL.so" : is a DataStreamer plugin
"libDataStreamMQTT.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
Number of plugins loaded:  8 

Loading compatible plugins from directory:  "/snap/plotjuggler/11/opt/ros/melodic/lib/plotjuggler_ros"
"libDataLoadROS.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Number of plugins loaded:  4 

Loading compatible plugins from directory:  "/home/batman/snap/plotjuggler/11/.local/share/PlotJuggler"
Number of plugins loaded:  0 
```

iejMac, 2021-02-23:
Solved by building the plugin with the same Qt version the snap was built with.
PlotJuggler has already made a huge difference in our development workflow on openpilot (https://github.com/commaai/openpilot). Let us know if  there is anything in the plugin that we can upstream.

closed 2021-02-23 by iejMac

facontidavide, 2021-05-13:
Hi,

I would love to discuss any option to upstream part of your work. Shall we continue this disucssion by email? davide.faconti@gmail.com