PlotJuggler/PlotJuggler #405: Exception from the plugin

banuc, 2021-02-12:
PloJuggler gave this error with .ulg file,

The plugin [DataLoad ULog] thrown the following exception:

ULog: error laoding definitions

Dou you have any suggestions?

copterspace, 2021-02-16:
David, this was reported and repaired already here: https://github.com/facontidavide/PlotJuggler/issues/170.
But now - same issue in  3.1.0 release...

facontidavide, 2021-02-16:
It can not be the same error, it is probably something else

banuc, 2021-02-22:
Yes btw im trying with 3.1.0 release.  I am able to upload .csv files but there is a problem with .ulg files.

facontidavide, 2021-02-24:
send me the incriminated file

copterspace, 2021-02-24:
For small example:
[log_29_2021-2-15-16-10-42.zip](https://github.com/facontidavide/PlotJuggler/files/6035207/log_29_2021-2-15-16-10-42.zip)

jphKelluu, 2021-02-24:
Hi. FYI same issue for me with .ulg file on plotJuggler 3.1 windows 10, but same .ulg file loads with plotJuggler 2.8.4.  Thanks for the great tool Davide!

facontidavide, 2021-02-24:
[image]

Looks good to me... are you experiencing issues on Windows?

copterspace, 2021-02-24:
Yes, on Windows 10 I get this:
[image]

banuc, 2021-02-25:
I also tried to upload with release 2.8.4 but i'm getting same error in windows 10.

[log_17_2021-1-29-17-36-16.zip](https://github.com/facontidavide/PlotJuggler/files/6041525/log_17_2021-1-29-17-36-16.zip)

facontidavide, 2021-02-25:
I will try to find some time during the weekend. But please keep in mind that maintaining the ULOG plugin is NOT even at the bottom of my TODO list.

facontidavide, 2021-02-27:
fixed

copterspace, 2021-05-18:
Installed recent https://github.com/facontidavide/PlotJuggler/releases/tag/3.1.1 - no change, same error...

jphKelluu, 2021-05-21:
following up on this issue, for me (windows10), Ulog data files now open in 3.1.1 with no problems! let the memes continue :)

facontidavide, 2021-05-21:
@copterspace , please provide the ulog file to reproduce the issue

copterspace, 2021-05-24:
Oh, guess I've found the reason:
In 3.1.1 the error  appeares, when I try to open the file from directory named: "C:\Users\ipetrov.BYTE-1C\Documents\QGroundControl\Logs\Альтитрал3\Полёты с гипером 2"
When I put the same file to "C:\Users\ipetrov.BYTE-1C\Documents\QGroundControl\Logs\" - it opens like a charm :)

In the same folder - once I rename it with Cyrylic symbols - the error appeares. Once I rename folder back with Latin symbols - file from this folder opens like a charm again. 
I had similar troubles with Opencv imwrite() function...

So problem seems to be in Cyrylic path interpretation on Windows.
Not sure you have time and wish to debug this. 
But this workaround might be mentioned in documentation...

closed 2021-05-25 by facontidavide

facontidavide, 2021-06-12:
Fixed