PlotJuggler/PlotJuggler #411: PlotJuggler missing messages when live streaming

JoshuaRiddell, 2021-02-24:
When streaming data from ROS, PlotJuggler seems to miss a message approximately every 1.3 seconds. The messages are a custom type sent at about 200Hz. I have a relatively new Xeon laptop so I don't expect it to be due to CPU load.

**Steps to reproduce:**
- unzip data file
- replay data file using "rosbag play plotjuggler_missing_message_test"
- connect plotjuggler to ROS and stream all messages
- output should look like the left screenshot
- to verify all messages are present in the bag file, the bag can be directly opened in PlotJuggler, it should look like the screenshot on the right

**Versions:**
Ubuntu 20.04 Desktop
ROS Noetic (apt install)
PlotJuggler 3.0.7 (snap install)

Thank you for your work on PlotJuggler. It's such an awesome tool and I love using it.

[image]

[plotjuggler_missing_message_test.zip](https://github.com/facontidavide/PlotJuggler/files/6032891/plotjuggler_missing_message_test.zip)

facontidavide, 2021-02-27:
No, it is not a plotjuggler problem, your data has that spikes.

This was taken looking at the rosbag and using the timestamp in Header

[image]

facontidavide, 2021-02-27:
If you think I am wrong, export the rosbag to csv and shgow me that the spikes are not there  ;)

closed 2021-02-27 by facontidavide