PlotJuggler/PlotJuggler #412: ULog: error loading definitions

mengchaoheng, 2021-03-02:
After the Motor Data in the File window, the following prompt appears：

[image]

mengchaoheng, 2021-03-02:
I have a px4 log file named log_2_2021-3-1-16-29-42.ulg and I want to view the data

mehmetyldz87, 2021-03-02:
Same problem 

File: PX4 [ULG File ]
[image]

Version: 3.1.0 
OS: Windows 10

bduva002, 2021-03-04:
I am also having an issue running 3.1 on win 10 with a ulog, mine just says not responding though.

facontidavide, 2021-05-18:
Please try version 3.1.1 and if the problem persist share the ulog file **as I ask every single time in any single issue**.

facontidavide, 2021-05-25:
Closed. Feel free to reopen once you can provide a way to reproduce (with a FILE)   with version 3.1.1.

closed 2021-05-25 by facontidavide