PlotJuggler/PlotJuggler #415: Resize and new tab buttons overlap

Georacer, 2021-03-18:
Hi there! Thanks a million for making PlotJuggler!

Here's a bug report for your troubles :P

Version: 3.0.7 from Snap store.

Issue: When top bar is full with tabs, the "new tab" button overlaps with the "Zoom all" button, making them hard to click.

Cheers!

P.S. If you could repackage releases in to .AppImages it would be great! Snap Store lags a bit.

facontidavide, 2021-04-01:
Hi,

Unfortunately when I work with Snaps instead of native builds, I do have problem like this that I have no idea how to address.
I spend a LOT of time to make AppImage work but eventually I had to give up.

Can you send me a screenshot?

Georacer, 2021-04-01:
[image]

I'm not opposed to building it myself, but some sort of packaging helps convince more people to use it tremendously. A private package repository would works just as well, providing a .deb package.

Georacer, 2021-04-02:
I have compiled from source in Ubuntu 20.04 and I have the same issue:
[image]

commit 5ae780be (2022-07-10): Fix #415
closed 2022-07-10 by facontidavide