PlotJuggler/PlotJuggler #417: Fails to build on Ubuntu 18.04

yangxiang92, 2021-03-23:
I fails to build latest master branch on Ubuntu 18.04 LTS.

I installed latest ros-melodic with the instructions on ros wiki: [ROS Wiki for Installization](http://wiki.ros.org/melodic/Installation/Ubuntu).

I also installed the binary mentioned in the README.md, which is:

```bash
sudo apt -y install qtbase5-dev libqt5svg5-dev libqt5websockets5-dev libqt5opengl5-dev
```

And after pulling the code into my catkin_ws and running catkin_make command, I got these errors:

```plain
-- No build type selected, default to RelWithDebInfo
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR)
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR)
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR)
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5X11Extras" with
  any of the following names:

    Qt5X11ExtrasConfig.cmake
    qt5x11extras-config.cmake

  Add the installation prefix of "Qt5X11Extras" to CMAKE_PREFIX_PATH or set
  "Qt5X11Extras_DIR" to a directory containing one of the above files.  If
  "Qt5X11Extras" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  PlotJuggler/3rdparty/Qt-Advanced-Docking/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/shawn/WorkSpace/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/shawn/WorkSpace/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

yangxiang92, 2021-03-23:
I fixed it after I installed `libqt5x11extras5-dev`.

facontidavide, 2021-04-01:
adding it to the readme

closed 2021-04-01 by facontidavide