PlotJuggler/PlotJuggler #419: Plot Jugger 3.10 on Windows 10 still problems reading log file

davidemiccone, 2021-03-25:
I attacched a ulg file that give me error "bad allocation" under Windows 10.

Any change to resolve?

Davide.

[09_01_28.zip](https://github.com/facontidavide/PlotJuggler/files/6203159/09_01_28.zip)

facontidavide, 2021-04-01:
Sorry to hear that. Unfortunately I have very little time to maintain ULog for the time being. I hope I will find some bandwidth in the future

dgeorge83616, 2021-04-15:
@davidemiccone, did you rebuild with this: https://github.com/facontidavide/PlotJuggler/commit/9505f51544518483d24e31ca9ea9135fa579fac1

facontidavide, 2021-05-18:
I just tried with the latest version and it work. Can you try version 3.1.1 that I released yesterday?

facontidavide, 2021-05-20:
Since I can not reproduce the issue with the given file, I am going to close this as "solved"

closed 2021-05-20 by facontidavide

facontidavide, 2021-06-12:
Fixed