PlotJuggler/PlotJuggler #421: Ardupilot binary log plugin

Georacer, 2021-04-11:
Hello Davide,

I'm interested in writing a plugin for parsing Ardupilot's binary logs, just as the PX4ulog plugin parses the PX4 logs.
I plan to use that as a "template" and work my way from there.

Are you willing to accept such a PR?
Do you have any early advice before I make any architectural decisions?

Thanks!

facontidavide, 2021-05-12:
Hi,

I decided that new plugins will be made available here: https://github.com/PlotJuggler in their own git repo.

In this way, you are responsible for maintaining it, document it, answer use issues, etc (I don't have bandwidth to maintain code written by someone else).

I created this as template: https://github.com/PlotJuggler/plotjuggler-sample-plugins

If you want to go ahead, let me know

closed 2021-05-12 by facontidavide

Georacer, 2021-05-12:
Ah, nice! Seems like a reasonable choice.

Some advice on good workflows/data flow and a guide for how they should be imported/loaded by the core Plotjuggler would be nice.

I won't start on this right now, so I'll let you know when, so that you can start the repo.

Thanks!