PlotJuggler/PlotJuggler #422: Ros2 Rolling release ?

doisyg, 2021-04-15:
Hi Davide,
Quick question, could you release plotjuggler for ros2 rolling ?
And if not, what would be the recommended way of installing plotjuggler for using it with ros2 rolling (precisely the verison that needs the most debugging) ? From source + ros plugins ?

doisyg, 2021-04-17:
Ok, so I managed to make it work with `rolling` by compiling from source the 3 repos:
https://github.com/PlotJuggler/plotjuggler_msgs.git
https://github.com/facontidavide/PlotJuggler.git
https://github.com/PlotJuggler/plotjuggler-ros-plugins.git

`plotjuggler_msgs` needs to be ros2 and I had to fork `plotjuggler-ros-plugins` to make it compile with rolling/galactic. PR is here: https://github.com/PlotJuggler/plotjuggler-ros-plugins/pull/11

Then starts and seems to work fine with `ros2 run plotjuggler plotjuggler`

salmagro, 2021-04-27:
Thank you very much for the implementation. Worked for me also in Rolling. But in my case, the executable is under `ros2 run plotjuggler PlotJuggler`.

I will replay some rosbags to check it.

salmagro, 2021-04-28:
I am receiving the following error when running it. Therefore I am not capable to subscribe to ROS2 topics. 
```
$ ~/ros2$ ros2 run plotjuggler PlotJuggler
Loading compatible plugins from directory:  "~/ros2/install/plotjuggler/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS2.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS2.so" :  "Cannot load library ~/ros2/install/plotjuggler/lib/plotjuggler/libDataStreamROS2.so: (~/ros2/install/plotjuggler/lib/plotjuggler/libDataStreamROS2.so: undefined symbol: _ZN17rosbag2_transport19GenericSubscription21handle_loaned_messageEPvRKN6rclcpp11MessageInfoE)"
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libTopicPublisherROS2.so" : is a StatePublisher plugin
Number of plugins loaded:  5 

Loading compatible plugins from directory:  "/usr/local/PlotJuggler/plugins"
Number of plugins loaded:  0 
```
Any recommendation about how to solve it?
Thank you very much.

doisyg, 2021-05-03:
> Which branches are you using?
A pending PR for `plotjuggler-ros-plugins` https://github.com/PlotJuggler/plotjuggler-ros-plugins/pull/11

This is what I get at lunch, I only tested opening bags, there is proably an issue with the libDataStreamROS2:

> $ ros2 run plotjuggler plotjuggler 
> Loading compatible plugins from directory:  "/home/gd/elodie2_ws/install/plotjuggler_ros/lib/plotjuggler_ros"
> "libDataLoadROS2.so" : is a DataLoader plugin
> "libDataStreamROS2.so" :  "Cannot load library /home/gd/elodie2_ws/build/plotjuggler_ros/bin/libDataStreamROS2.so: (/home/gd/elodie2_ws/build/plotjuggler_ros/bin/libDataStreamROS2.so: undefined symbol: _ZN17rosbag2_transport19GenericSubscription21handle_loaned_messageEPvRKN6rclcpp11MessageInfoE)"
> "libTopicPublisherROS2.so" : is a StatePublisher plugin
> Number of plugins loaded:  2 
> 
> Loading compatible plugins from directory:  "/home/gd/elodie2_ws/build/plotjuggler/bin"
> "libDataLoadCSV.so" : is a DataLoader plugin
> "libDataLoadULog.so" : is a DataLoader plugin
> "libDataStreamLSL.so" : is a DataStreamer plugin
> "libDataStreamMQTT.so" : is a DataStreamer plugin
> "libDataStreamSample.so" : is a DataStreamer plugin
> "libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
> "libDataStreamUDP.so" : is a DataStreamer plugin
> "libDataStreamWebSocket.so" : is a DataStreamer plugin
> "libDataStreamZMQ.so" : is a DataStreamer plugin
> Number of plugins loaded:  8 
> 
> Loading compatible plugins from directory:  "/home/gd/.local/share/PlotJuggler"
> Number of plugins loaded:  0 

I will try to fix it if Davide doesn't do it soon

facontidavide, 2021-05-16:
I pushed the PR for the rolling distro of **plotjuggler** and **plotjuggler_msgs**.

I will wait for your green light before pushing the **plotjuggler_ros_plugin**

doisyg, 2021-05-17:
Great, I won't have time however to fix the **libDataStreamROS2** plugin in **plotjuggler_ros_plugin** issue before next month, so if you have something already, go ahead

closed 2026-01-06 by facontidavide