PlotJuggler/PlotJuggler #426: Help needed compiling for macOS BigSur (missing -lrt error)

bertaveira, 2021-04-25:
Hi! This might be a dumb cry for help. Sorry if it is but I can't find help anywhere about my specific error compiling on macOS BigSur (non M1).

I have installed command line developer tools and homebrew. Then using brew I installed qt5 and cmake:

```bash
brew install qt@5
brew install cmake
```

Then I cloned this repo and verified I was in the main. I then followed the instructions:

```bash
mkdir build; cd build
cmake ..
make
```

This last make command fails at 76% saying that it cant find the library `-lrt` that from what I understand and find online is a library that is not supported in macOS for realtime applications and usually ignored in macOS.

My question is how do I remove this flag or make it work?

The ending of the terminal output for the `make` command is the following:

```bash
[ 21%] Automatic MOC and UIC for target DataStreamUDP
[ 27%] Built target qt_advanced_docking
[ 35%] Built target common_obj_static
[ 35%] Built target lslboost
[ 35%] Built target paho-mqtt3a-static
[ 36%] Built target paho-mqtt3c-static
[ 36%] Automatic MOC and UIC for target lslobj
[ 36%] Automatic MOC and UIC for target DataStreamMQTT
[ 36%] Built target lslobj_autogen
[ 36%] Built target DataStreamMQTT_autogen
[ 64%] Built target qwt_static
[ 64%] Built target DataStreamMQTT
[ 65%] Automatic MOC and UIC for target plotjuggler
[ 74%] Built target lslobj
[ 74%] Automatic MOC and UIC for target lsl-static
[ 74%] Built target lsl-static_autogen
[ 75%] Built target lsl-static
[ 75%] Automatic MOC and UIC for target DataStreamLSL
[ 75%] Built target DataStreamLSL_autogen
[ 76%] Linking CXX shared library ../../bin/libDataStreamLSL.dylib
ld: library not found for -lrt
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/libDataStreamLSL.dylib] Error 1
make[1]: *** [plugins/DataStreamLSL/CMakeFiles/DataStreamLSL.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 76%] Built target DataStreamSample_autogen
[ 76%] Built target DataLoadULog_autogen
[ 76%] Built target DataLoadCSV_autogen
[ 76%] Built target DataStreamUDP_autogen
[ 76%] Built target DataStreamWebSocket_autogen
[ 76%] Built target plotjuggler_autogen
make: *** [all] Error 2
```

bertaveira, 2021-04-26:
Ok I have found the place were it asks for the -lrt (rt). I removed the flag and it works now. I don't know if there is a work around it more elegant but it was on `3rdparty/liblsl/CMakeLists.txt`

Here is the diff:

```bash
git diff CMakeLists.txt 
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 747b14a2..1d458d2e 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -149,7 +149,7 @@ if(UNIX)
                 ${CMAKE_CURRENT_SOURCE_DIR}/lslboost)
         target_link_libraries(lslobj
                 PRIVATE lslboost
-                PUBLIC dl rt )
+                PUBLIC dl )
 endif()
```

As you can see I just removed the `rt`.

Hope it helps someone... Feel free to indicate a better way of doing it besides changing this

jsandleraol, 2021-05-24:
@bertaveira, how do you run the program afterward? I cannot find the executable.