PlotJuggler/PlotJuggler #428: Crash after UDP connect

bertaveira, 2021-04-26:
Hi! I have compiled, installed and it is running on macOS Big Sur (non M1). However despite it opening the app by typing `plotjuggler` on the terminal when I connect to a UDP port and then try to drag a signal to plot, it just crashes.

Here is the terminal output:

```bash
UDP listening on port 9870
Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
QOpenGLEngineShaderManager::useCorrectShaderProg() - Qt::NoBrush style is set
Stack trace (most recent call last):
#2    Object "libsystem_platform.dylib", at 0x7fff2037ad7c, in _sigtramp + 28
#1    Object "plotjuggler", at 0x10322bfbd, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#0    Object "plotjuggler", at 0x10322c022, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 66
zsh: abort      plotjuggler
```

Hope someone also with a mac can help or someone who knows this error.

facontidavide, 2021-05-18:
Is there anyway I can reproduce this?

bertaveira, 2021-05-20:
I don't have sorry. But any mac (not M1) this seems to happen. We tried in a few (each compiling themselves) and all of them crashed. Haven't tried on an M1 because I don't have access to one but intel ones this happens. On a MacOS virtual machine this seems to also happen maybe a way for you to reproduce this

facontidavide, 2021-07-11:
It is now clear that PlotJuggler is broken on MacOS and not having one, I don't see this changing any time soon.

Sorry.

closed 2021-07-11 by facontidavide