PlotJuggler/PlotJuggler #430: symbol lookup error using ros-melodic-plotjuggler

dc217863, 2021-05-08:
Hi,

I have installed PlotJuggler using `sudo apt install ros-melodic-plotjuggler-ros` which installs 'ros-melodic-plotjuggler' also. I am using ros-melodic on Ubuntu 18.04.

With this installation, I am unable to load any rosbags into plotjuggler. On doing so, I get the following error:

> /opt/ros/melodic/lib/plotjuggler/plotjuggler: symbol lookup error: /opt/ros/melodic/lib//librosbag_storage_plugins.so: undefined symbol: _ZN6rosbag6Buffer7setSizeEj

I created also a minumum example of rosbag using the example in http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data and face the same issue.

Any help would be most appreciated!

facontidavide, 2021-05-12:
I checked my Melodic installation and it seems to work just fine. Are you launching it with

      rosrun plotjuggler plotjuggler

Can you try this too?

      sudo apt update
      sudo apt upgrade

dc217863, 2021-05-12:
Thanks for having a look. Upgrading 'ros-melodic-rosbag-storage' solved this issue. I still had issues in actually loading the rosbag file which I was analyzing. 

My solution was to use an older version (2.8.4).

closed 2021-05-12 by dc217863

facontidavide, 2021-05-12:
If you can share the rosbag with me,  I will have a look