PlotJuggler/PlotJuggler #431: Forward slash in CSV column name results in child items in Timeseries list

seanmcleod, 2021-05-08:
Column names in CSV files often include the units as part of the column name and so they often have forward slashes, e.g. **Vertical Speed (ft/min)**.

Which then results in the following in the **Timeseries List**.

[image]

Which is a bit messy visually, but also results in more work when trying to drag that particular column onto the graph, since you first have to expand it and then drag the child item **min** in this example onto the graph.

facontidavide, 2021-05-10:
This is a built in characteristic of PlotJuggler.

I guess the best option would be to change the separator character in the preference menu.

seanmcleod, 2021-05-10:
If I open **App -> Preferences...** I don't see any mention of a **separator character** option in either tab. I'm running version 3.1 from the Windows installer link from Readme.md.

[image]

facontidavide, 2021-05-10:
yes, I know. What I meant is that this option need to be developed, because I don't have it yet.

seanmcleod, 2021-05-10:
Ah, okay got it. I thought you were referring to an an existing option in the current preferences dialog.

seanmcleod, 2021-05-10:
I assumed the forward slash separator was designed specifically for data in a hierarchical format, e.g. json encoded data etc. and so shouldn't be applied for CSV data.

facontidavide, 2021-05-11:
I introduced a quick solution, that is to disable entirely the tree view and the "/" separator.

closed 2021-05-11 by facontidavide