PlotJuggler/PlotJuggler #445: How to show original timestamps in plot view, not received time.

Tumiz, 2021-05-19:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
I use UDP to feed plotjuggler data with timestamps. How to show the timestamps on chart's x axis?

# Steps to reproduce (important)
run the following script
```
import socket
import math
import json
from time import sleep

sock = socket.socket(socket.AF_INET, # Internet
                     socket.SOCK_DGRAM) # UDP
time = 0.0

while True:
    sleep(0.05)
    time += 10

    data = {
        "timestamp": time,
        "test_data": {
            "cos": math.cos(time),
            "sin": math.sin(time)
        }
    }
    sock.sendto( json.dumps(data), ("127.0.0.1", 9870) )
```

Tumiz, 2021-05-19:
[image]

facontidavide, 2021-05-19:
Fixed! Thanks for reporting

commit 0be55209 (2021-05-19): fix #445
closed 2021-05-19 by facontidavide