PlotJuggler/PlotJuggler #446: Missing .so in 3.1.1 linux installer

russkel, 2021-05-21:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
Need to include libdouble-conversion.so.1 in the Linux installer binary.

```
↳ ./plotjuggler/plotjuggler.sh
/home/russ/plotjuggler/plotjuggler/bin/plotjuggler: error while loading shared libraries: libdouble-conversion.so.1: cannot open shared object file: No such file or directory
```

# Steps to reproduce (important)

- Manjaro Linux
- Install via 3.1.1 linux installer .run

russkel, 2021-05-21:
Running the following in the plotjuggler/lib directory did seem to 'fix' the error and run PlotJuggler. Still this is a bit of hack.
```
ln -s /lib/libdouble-conversion.so libdouble-conversion.so.1
```

facontidavide, 2021-05-21:
yes, I realized that doesn't work as well as I want to. I need to iterate more to make this really useful.

I think eventually I need to come back to AppImage instead

facontidavide, 2021-05-25:
Removed that broken installer. Try the AppImage https://github.com/facontidavide/PlotJuggler/releases/tag/3.1.1

closed 2021-05-25 by facontidavide