PlotJuggler/PlotJuggler #449: New XY plot stuck in loop

broughtong, 2021-05-29:
I listen to a message either from rosbag or stream, and then try to do an xy plot of two of the fields. 

I get a window asking if i want to swap the axes or rename it. If click ok, the box disappears and reappears. Same thing again and again.

The platform is ros noetic ubuntu 20.04 i3

also happens with the dummy data -t. Click _cos, _sin. Right click drag to plot. Window for the X and Y swap and custom name reopens everytime I click ok. Then need to sigterm to kill the program.

facontidavide, 2021-05-29:
Can you confirm the version? 3.1.1?

broughtong, 2021-05-29:
Yes currently 3.1.1, but I had it before in previous versions too, so I dont think its a new problem.

facontidavide, 2021-05-29:
I am sorry that you are experiencing this issue. 

I really want to help you and solve this, but I am not able to reproduce it on my machine :cry:

broughtong, 2021-05-30:
Do you have any suggestions as to where I could look into the code to find the problem?

broughtong, 2021-06-04:
It seems to be linked to these lines of code (plotwidget.cpp, line 444): 
```
while (name.empty())
{
   SuggestDialog dialog(name_x, name_y, this);
   bool ok = (dialog.exec() == QDialog::Accepted); //this line triggers a continue/jump back to the while line when the user clicks 'ok'
```
Honestly I have no idea how a function call is causing the code to continue/jump back to the loop.
I had a look at QT dialog exec function, and they warn:

> Note: Avoid using this function; instead, use open(). Unlike exec(), open() is asynchronous, and does not spin an additional event loop. This prevents a series of dangerous bugs from happening (e.g. deleting the dialog's parent while the dialog is open via exec()). When using open() you can connect to the finished() signal of QDialog to be notified when the dialog is closed.

Could this be the cause?

facontidavide, 2021-06-09:
>SuggestDialog 

Unlikely. I will investigate, but I don't understand why this never happened to me.
Are you compiling or using the debian?

If you are compiling, which version of Qt are you using?

broughtong, 2021-06-13:
The version of Qt is 5.12.8.
I tried with both the ubuntu repository version and compiled from source, same problem.

brunopinto900, 2021-06-16:
Hello, the error happens when you press the OK button, but when you press ENTER two times, it doesn't. I dunno.

broughtong, 2021-06-17:
I tried it, but for me it happens regardless of number of times pressing enter/clicking ok, or various combinations

broughtong, 2021-08-20:
It would seem that this happens under the i3 desktop, but if I switch to the stock ubuntu desktop the problem disappears. Otherwise running the same OS

dzid26, 2021-08-21:
I am on WSL2 with VcXsrv using OpenGL.  Qt is 5.12.8.
It's looping and the only way to stop the loop is by clicking X on Windows' toolbar.  
Then PlotJuggler plots several curves with different names depending on how many times Ok was pressed after typing a name.

facontidavide, 2026-09-20:
The XY-curve creation dialog was rebuilt from scratch in PlotJuggler 4 on Qt 6.11 (XYCurveDialog), and the original loop was tied to a Qt 5 / window-manager interaction (i3, WSL2/VcXsrv software GL) that reporters found vanished under a normal desktop. Closing as obsolete on the new stack; please reopen against PJ4 if you ever see the dialog re-appear.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide