PlotJuggler/PlotJuggler #453: v3.1.1 ULOG: Failed to open replay file on Windows

mehmetyldz87, 2021-06-07:
# Problem description
The ulog file given below is not opening in v3.1.1 on Windows
[image]

# Steps to reproduce (important)

- Windows 10 
- ULog File 
https://drive.google.com/file/d/1C93OcuBA2RcVH-z3zryoc-g1ogwZzuy9/view?usp=sharing

Thanks for your time and support.

facontidavide, 2021-06-09:
I will close the issue. If you have any additional information about how to reproduce this, feel free to open the issue again.

closed 2021-06-09 by facontidavide

mehmetyldz87, 2021-06-09:
Hi @facontidavide 
I think the problem is caused by the name or location of the folder containing the ulog. It opens on desktop or downloads folder .    
Thanks for your time and support.

facontidavide, 2021-06-12:
Fixed

hauserulrich, 2022-11-05:
Hi @facontidavide,
I am on 3.6.0, Windows 10. When I Click on the Button for Data, the only options I get are csv, mcap and ulg. My PX4-files are bin or log. I cannot see or open them. Do I need to install a PX4 plugin seperately? I see a directory for a ulog plugin, but I cannot see any compiled plugins or a way to create them without all the cloning and compiling effort. Is there an easy solution in windos? Dragging a bin file into the window didn't lead to any complaints but there was no display of data either. Dragging a log file asked for a loader and ended in an error. So something must happen while dragging and dropping.