PlotJuggler/PlotJuggler #457: Migrating LSL plugin

facontidavide, 2021-06-12:
Hi
    Celal Savur (@csavur)
    Shitij Kumar (@spk4422)
    Mazin Ali (@mazinms)

With the potentially growing number of plugins, I would like to start migrating some to their own repository, in this case

**plotjuggler/plotjuggler-lsl**

Ideally, we can add this as an optional git submodule that this app may download. This will also allow you to maintain this plugin more easily, since it will have it's own repo / issues / readme.

I will start moving the MQTT plugin, in this way you may see how we integrate main app with separated plugin. Is that ok For you?

csavur, 2021-06-14:
Hello Davide,

It make sense. Please let me know if you need any help during migration.

Best regards,
Celal

-- 
*Celal Savur,*
*Ph.D. Candidate, PhD. in Engineering Program*
*Program co-chair, SOSE 2020*
*Rochester Institute of Technology*

Email: ***@***.***

*CONFIDENTIALITY NOTE*: The information transmitted, including attachments,
is intended only for the person or entity to which it is addressed and may
contain confidential and/or privileged material. Any review,
retransmission, dissemination or other use of, or taking of any action in
reliance upon, this information by persons or entities other than the
intended recipient is prohibited. If you received this in error, please
contact the sender and destroy any copies of this document.

csavur, 2021-06-14:
It is just an idea. Can we make plugins like QtCreator plugins. We can enable and disable from the settings? In this way, anyone wants to use the lsl plugin can be enabled or disabled?

facontidavide, 2021-06-16:
That would be nice. coming soon hopefully. I made the repo: https://github.com/PlotJuggler/plotjuggler-lsl

closed 2021-06-16 by facontidavide