PlotJuggler/PlotJuggler #458: Message to large?

lgulich, 2021-06-14:
Hi, first of all thank you for this great tool! It really is a game changer.

I am facing a problem when trying to load a custom message from a rosbag. After opening a rosbag and selecting the topic the following exception occurs:

>  The plugin [DataLoad ROS bags] thrown the following exception: 
   Buffer overrun in RosIntrospection::ReadFromBuffer

I am using plotjuggler 3.1.2 with ros-melodic on ubuntu-18.04
The used rosbag is the following (only contains a single topic): 
[2021-06-14-09-48-08.zip](https://github.com/facontidavide/PlotJuggler/files/6646550/2021-06-14-09-48-08.zip)

Any ideas how this can be solved?

facontidavide, 2021-06-14:
I will have a look.

lgulich, 2021-07-05:
@facontidavide do you have any idea what this could be? I am also happy to try to fix this myself but I would be glad about some pointers in the right direction.

facontidavide, 2021-07-05:
I can see that there is something wrong in the message definition.

>"# All readings of the Ascento robot.
>sensor_msgs/Imu imu_measurement
>encoder_tools_ros/MeasurementMe"

The definition of encoder_tools_ros is missing and PlotJuggler can't understand this message. 

This is a known problem when **rosbridge** is used. Is this your case?

lgulich, 2021-07-06:
Interesting, no we do not use rosbridge. I will investigate this wrong message definition, thanks!

lgulich, 2021-07-06:
How did you extract the message definition from the bag? I just tried it with the script from [here](https://answers.ros.org/question/230120/extracting-definition-of-customized-messages-from-bag-files/) and got the correct message definition:

> \# All readings of the Ascento robot.
sensor_msgs/Imu imu_measurement
encoder_tools_ros/MeasurementMessage[2] hip_encoder_measurements
rotary_actuator_ros/ReadingMessage[2] leg_readings
rotary_actuator_ros/ReadingMessage[2] wheel_readings

facontidavide, 2026-09-20:
This was the old ROS1 .bag loader (RosIntrospection) choking on a bag with a truncated custom message definition — a recording-side defect the maintainer already diagnosed. PlotJuggler 4 drops the ROS1 bag path entirely (ROS data flows through MCAP and live ROS2 via rosx_introspection), so this specific failure no longer exists; closing as obsolete.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide