PlotJuggler/PlotJuggler #463: Y tick labels lose precision when data magnitudes are small

sjs329, 2021-06-18:
# Problem description

In a recent version update of PlotJuggler (within the last year...) the behavior of Y-axis tick labels seems to have changed. When the magnitude of the data in view is small enough, the significant figures / precision of the labels is no longer enough to be useful. See attached image where the Y axis labels are all zeros. This makes it much harder to tell what's happening with signals that are significant at very small magnitudes.
[image]

# Steps to reproduce (important)

- Describe your platform / Operative System:
-- Ubuntu 18.04, PlotJuggler version 3.2.0, using ROS plugin (though that doesn't seem to be necessary to reproduce)
- CSV file to reproduce: 
[small_magnitude_example_modified.csv](https://github.com/facontidavide/PlotJuggler/files/6677907/small_magnitude_example_modified.csv)

facontidavide, 2021-06-18:
fixed

closed 2021-06-18 by facontidavide

sjs329, 2021-06-18:
Fixed indeed! Thanks @facontidavide !
[image]