PlotJuggler/PlotJuggler #464: [Bug] Does not load new data when using the -l and -d flag (would be good for automatic imagefile creation)

mistermult, 2021-06-18:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

If the program is started with the flag '-l layoutfile.xml -d rosbag.bag' the new data is displayed. Instead the old data from the new run is displayed. Bug is in the Version for ROS kinetic and ROS melodic.

# Steps to reproduce (important)

- Operative System: Ubuntu 18.04 inside Docker or outside
- Steps:
  - Open Plotjuggler
  - Open a bag file 'bag1.bag'
  - Create some plots
  - Save the layout to layout.xml
  - Close the program
  - Open the program from commandline with `rosrun plotjuggler PlotJuggler -d bag2.bag -l layout.xml`
  - Press Ctrl+A to import everything. Press OK. You see a progress bar for the import.
  - Confirm that you want to replace the old data.
  - Another progress bar for loading data appears.
  - Plots with data from`'bag1.bag` is shown.

# Background
I have done the testing manually (my own hands and monitor). In the end, i want to automatically create plots from files from a given layout. All files have the same ROS topics (but might have different length of recordings). More advanced, I want to do it in the CI pipeline. Even though, plotjuggler is a GUI tool, I managed to do it:
* by sending keys to xdotool 
* by using xvfb when being on the headless CI server
However, plotjuggler does not display the data of the new file, so it does not work.

# PossibIe Improvements
1) Fix the bug described above --> Then we can use it for automatic creation of plots, but plots it is cumbersome and brittle.
2) To easily make a screenshot automatically after plotjuggler has loaded, make a flag so that plojuggler loads without needing clicks/keystrokes. The flag should:
  - Load all topics from the layoutfile (or just all)
  - Display the bag with the given layout
3) However, in this way we can only create a screenshot of the first tab automatically. 
 - The easiest way would be to add a shortcut to change tabs (e.g. TAB). Then we can send TAB to plotjuggler and make a screenshot of the next. This would also be good for humans.
4) Maybe, there should be a option to remove the left topic component or make it smaller. It is not saved in the layoutfile. So space is wasted.
5) Ideally, there should be a command line option to create plots from files, but I guess this more involved.

- If 1) is done, it is possible to make images automatically in a brittle way. If 2) is done, it would be less brittle and used reliable.
- If 3) is done, layout files with multiple tabs (e.g. the original ones) can be reused. Otherwise, one has to manage multiple copies.
- 4 and 5 are just the icing on the cake but not really neeeded.

mistermult, 2021-06-18:
I guess you and other might be interested in creating automatic screenshots. This might be relevant to #137 and #362. I do in the following way:
```bash
rosrun plotjuggler PlotJuggler -d bagfile.bag -l layout_rosbag.xml &
sleep 20
WINDOWID=$(xdotool search --class PlotJuggler)
xdotool windowactivate ${WINDOWID} && xdotool key ctrl+a && xdotool key ctrl+a && xdotool key Return && xdotool key Return
sleep 60
xdotool windowactivate ${WINDOWID} && xdotool key Return && xdotool key Return
```
Now plotjuggler should be loaded if there was no bug. Then, do a screenshot and end it:
```
scrot
killall -9 plotjuggler
```
If you want do do it headless, then do the following
```bash
Xvfb :2 -screen 0 1024x768x24 -fbdir /var/tmp 
export DISPLAY=:2
# Do commands as above but make the screenshot instead of scrot with
xwd -root -silent | convert xwd:- png:/tmp/screenshot.png
```

facontidavide, 2021-06-18:
Please send me your lauyout.xml, I think I know what is going on

facontidavide, 2021-06-18:
For the records, I will not work on creating this "automatic images" workflow, but I think you found a bug anyway that need to be solved.